gdombiak/OctoPrint-OctoPod #1: Notification Server URL Help

Morcegolas, 2019-05-28:
Hello, first of all thanks for you hard work. I use Octopod in my iPhone to check the printer while I’m not home. Today I updated your app and installed the plugin in octoprint that is running in my RaspberryPI connected to my printer. I tried the Send test notification but still have in the Notification Server URL: http://octopodprint.com/ what I assume that is wrong and it didn’t work. My question is what should go in there? http://localhost ?

Thanks again.

taker218, 2019-05-28:
I also have the same problem.
Installed the plugin and after I connected to my OctoPi via app I could send a notification, but I didn't get one.

Put in http://localhost and this crashed my OctoPrint-Instance when I tried to send the test notification.

pjratl, 2019-05-28:
Same issue ....

taker218, 2019-05-28:
I just tested it again and once it worked.

I now have enabled debug logging and get this in the octoprint.log:

> 2019-05-28 19:44:09,970 - octoprint.plugins.octopod - DEBUG - Snapshot URL: http://localhost:8080/?action=snapshot
> 2019-05-28 19:44:11,673 - octoprint.plugins.octopod - DEBUG - Print Job Notification Response code: 204
> 2019-05-28 19:44:13,126 - octoprint.plugins.octopod - DEBUG - CheckTemps(): {'chamber': {'actual': None, 'target': None, 'offset': 0}, 'bed': {'actual': None, 'target': None, 'offset': 0}, 'tool0': {'actual': 0.0, 'target': None, 'offset': 0}}

pjratl, 2019-05-28:
working now with default values. I'm going to assume it was a server issue that's been resolved

taker218, 2019-05-28:
Not working on my end here.
At least not, when I have the app closed completely (not in task list of iOS visible).

If I close the app down (what I do most of the time) I get no notification at all (tried it with the bed heating notification).
When I have the app open, but in the background (in task list of iOS visible) I get the notification

Guilouz, 2019-05-28:
Same here, octopod must be running in background to have notifications, not working when it's closed.

gdombiak, 2019-05-29:
Hey guys,

Thanks for using the brand new plugin and providing this useful feedback. Until I get to the bottom of this, make sure to leave the app running in the background. It seems like silent notifications (which is what we use) are treated differently by the iOS when the app is not running. I tested many things but not having the app closed. :(

Until this gets resolved, please let me know how it goes by leaving the app in the background.

Leave the server url as "http://octopodprint.com/". That is the correct address. What you might want to change is the URL to the camera snapshot. The default should work with OctoPi out of the box.

gdombiak, 2019-05-29:
Looking at [APNS Silent Push Notifications documentation](https://developer.apple.com/documentation/uikit/uiapplicationdelegate/1623013-application) we can see that if the user forced-quit (i.e. killed) the app then silent notifications are not sent since the app is not running. This is how Apple let users control unwanted notifications by letting them kill the app.

IOW, not much we can do unless we do not use silent notifications which is a bigger change since translations will need to be stored in the plugin and not in the iOS app. Same for printer name that user gave in the iOS app.

TLDR; :(

jollino, 2019-06-18:
My vote is for 'regular' notifications as well, as it would have interesting side effects.
https://developer.apple.com/library/archive/documentation/NetworkingInternet/Conceptual/RemoteNotificationsPG/CreatingtheNotificationPayload.html shows a way to localize notification payloads by keeping the translations in the app bundle and storing the language with the server, so that the notification payload contains the language to use as a lookup key.

However, in my humble opinion it may actually make sense to keep everything on the server (by "server" I mean "the server that currently handle notifications for users").
Look at it this way: when it comes to notifications, the master is really the OctoPrint plugin; OctoPod on iOS is really just a receiver. Indeed, when a notification is sent from the server, OctoPod just displays it and doesn't really do anything else.

Right now if you were to add more notification hooks to the plugin ("notify the user when the printer reaches a certain percentage" or "notify the user every X hours" or anything else), you would have to also modify OctoPod and rebuild it just so it can handle the new message; end users would have to wait until a new release is approved by Apple.

By decoupling the server from the iOS app, so that OctoPod can just show whatever the server sends, would allow new notification hooks to be added asynchronously. All it has to do is really just show the notification, and the language issue can be handled by simply saving the language itself together with the token (maybe on each app load in case the user changes it); then the server can locally look up the appropriate language for the string being requested by OctoPrint for that specific event, and put it right into the APNS envelope from the get-go.

Bonus side effects: works with the app closed (iOS may kill it, etc.) and doesn't require any work at all on iOS when it comes (except generating an alert to show it in case it comes when the app is open).

Just my two cents, and of course I may be missing something huge here as I haven't written code for iOS since iOS 4. :)

gdombiak, 2019-06-25:
Major work is almost done! I will soon release new beta version of OctoPod that will require the new version of this plugin. You will get visual alerts even if you killed the app. On top of that, I added new alerts for any firmware error (e.g. probing failed, runout temp error, min temp error, max temp error, etc).

jollino, 2019-06-25:
Sweet! Thanks Gaston! :)

gdombiak, 2019-06-30:
New version is now out and includes a fix for what was raised in this ticket. Thanks for bringing this up. 

Closing ticket now.

closed 2019-06-30 by gdombiak