gdombiak/OctoPrint-OctoPod #12: Web camera wont connect

Cesar0415, 2019-07-06:
Request Error HTTP response 404

gdombiak, 2019-07-06:
Hi Cesar,

The camera URL (address) seems to be pointing to the wrong place. What is the URL it is trying to reach? Are you on the same network?

gdombiak, 2019-07-06:
Have you tried manually copying the camera URL and open that address on the browser? The default value works for people using octopi

Cesar0415, 2019-07-06:
When I type it in manually it doesn’t seem to work but when I test it on
the octopi interface it’s fine

gdombiak, 2019-07-07:
Cesar, hablas español?

What do you get when you paste the camera URL in the browser?

What does it mean to test on the octopi interface? How are you testing things?

gdombiak, 2019-08-11:
Cesar, I'm closing this ticket due to inactivity. We can open a new one if you are still running into this issue.

Thanks,
Gaston

closed 2019-08-11 by gdombiak