gdombiak/OctoPrint-OctoPod #21: Connect from anywhere

ZeekEteek, 2019-09-08:
Is there any way to connect using the app from not on my home network?

gdombiak, 2019-09-09:
Hi @ZeekEteek, you can absolutely use OctoPod iOS app outside of your home network. As you noticed this [plugin](https://github.com/gdombiak/OctoPrint-OctoPod) will send you notifications that also work outside of your home network. To make the iOS OctoPod app work outside of your network you can follow this [VPN Setup guide](https://github.com/gdombiak/OctoPod/wiki/Using-OpenVPN-for-Remote-Access). That is the most secure way to go. A less secure way to go is to use a reverse proxy with authentication. It might be easier to setup but less secure compared to a VPN. Both work equally fine.

Gaston

GS-BS, 2019-09-17:
I cant use VPN so I  must need a subdomain to connect Octoprint. The webinterface works well but the app give me an error: Starscream.WSError 1
Maybe I need the ports to make it work.

gdombiak, 2019-09-17:
If you cannot use a VPN then checkout these 2 guides for setting up a port-forward+reverse proxy+authentication.

https://printoid.net/access-octoprint-from-the-internet/
https://blog.dembowski.net/2018/octoprint-and-reverse-proxies/

Gaston

gdombiak, 2019-10-19:
Closing ticket for now. Open a new one if still need help

closed 2019-10-19 by gdombiak