gdombiak/OctoPrint-OctoPod #25: Push Notification Server - Security Risk / Self Hosting

natiz, 2019-10-20:
Hi,

I appreciate all the work around this plugin, I love it!

I do have a concern regarding the push notifications. It seems that push notifications from OctoPrint go to "http://octopodprint.com/" (which seems to be configurable), using the non-secure HTTP protocol, without any authentication/authorization in place, which poses a risk of exploiting the PN token.

1) Are there plans to provide extra layers of security around it? starting with the basic HTTPS connection, maybe a TLS as well?
2) Can one implement its own self-hosted PN server? I believe that would require you to provide the APNS TLS keys/certificates, not sure if that's ideal, do you know any way around it?
3) In the spirit of open source, can you publish the code to the push notification server? certificates/keys should be omitted of course.

BTW, this is how a push looks like to octopodprint.com
```
POST //v1/push_printer HTTP/1.1
HOST: octopodprint.com
Connection: keep-alive
Accept-Encoding: gzip, deflate
Accept: */*
User-Agent: python-requests/2.21.0
Content-Length: 208
Content-Type: application/json

{"sound": "default", "useDev": false, "printerName": "OctoPrint", "title": "OctoPrint", "tokens": ["XXXXXXXXXXXZZZZZZXXXXX"], "message": "Print complete"}
```

gdombiak, 2019-10-21:
Shalom Nati,

Eij ata? :) Would you mind sending me an email with any vulnerability that you found (e.g. being configurable)? 

Some contextual information: I spend several months looking for a sponsor to help pay the hosting cost with no luck. I really wanted to share this functionality with the community so I decided to pay the monthly cost myself. I purchased the domain and also the SSL certificate. The service is being hosted in Lightsail (AWS) using the smallest/cheapest plan. It is so small that I'm not sure that it can handle the load and do TLS negotiation. :( This is why I never installed the SSL/TLS certificate to do HTTPS. Therefore, traffic is not encrypted as you noticed. Considering the worst that could happen if hacked (that I could imagine), given the fact that there is no personal information stored or transmitted, it could be that people can get spammed with notifications that are being injected from a hacker. If I can find a sponsor for hosting the service then I can use a bigger virtual server and enabled HTTPS so traffic is protected and this vector of attack is gone.

The source code is available but not public to NOT provide any information that a hacker might use to their advantage. As you correctly guessed, to run your own notifications service and still use the released iOS app, you will need to have access to my developers credentials. That is not an option. However, if you want to release your own iOS app using the same source code then I can share the source code of the service so you can use your own developers credentials.

Toda,
Gaston (ani jashafti ibrit sheli)

natiz, 2019-10-21:
Hi Gaston, thanks addressing my concern!

By configurable I mean that you can change the value of the push URL in OctoPrint. The vulnerable part here is the traffic in HTTP. While it's not major, as you mentioned, it can still be avoided.
So setting up HTTPS would be a good start. [LetsEncrypt](https://letsencrypt.org/) can be used to issue a certificate for free, it is a free/open CA that provides an agent you install that also does the renewals. There are a [couple of tutorials I found](https://lightsail.aws.amazon.com/ls/docs/en_us/search?s=let%27s%20encrypt) for Lightsail.

In addition, I recommend looking into other AWS services that could achieve the same goal (note that I'm no AWS expert), and could potentially reduce costs:
1) Using a [Lambda](https://aws.amazon.com/lambda/pricing/) function rather than running a hosted server
2) Using [AWS SNS](https://aws.amazon.com/sns/), not sure what the traffic is, but they have 1 million notifications/month free for the free tier

Let me know if you need help setting things up, will be happy to help

jollino, 2019-10-21:
Gaston,
Lightsail's most basic plan seems pretty decent actually, I have a similar VPS (on a much sketchier ISP :D) and while I only run a few small python bots on it now, I did run a full-featured Apache on it with a few vhosts some time ago. As long as it's not being killed with traffic, it's just fine.

It looks from @natiz 's post that the server is really a bare-bones python script (does it use WSGI at all?), so the easiest approach would be to move it to a different port on localhost only, and install nginx with an SSL certificate on port 443, forwarding everything to the actual python server via localhost. I never configured nginx myself but it looks like a pretty simple thing: https://docs.nginx.com/nginx/admin-guide/web-server/reverse-proxy/
(Of course you may want to keep it also on port 80 so current/older version of the app don't stop working, while newer ones can use ssl on 443.)

SNS (or other similar platforms running notifications for apps) may indeed be the best approach to get rid of the intermediate server entirely and make it simple, however I was thinking that it may end up being limited later on: say we want to allow users to upload the image, etc.; having something in between where the data can be massaged can be useful, and it wouldn't be limited by the time-sharing nature of Lambda/serverless either (at this stage it may actually be overkill as we're just sending a notification, but the moment we do something else, those 100 ms time chunks are going to add up!)

My personal vote is to keep Lightsail, or any other similar VPS, and add nginx in front of it to both make it sturdier (it's extremely efficient!) *and* handle SSL. It shouldn't be too disruptive as a change, indeed it would be completely transparent to current clients.

Moreover, you could also add Cloudflare (the free plan) to further protect the server from any DDoS; actual requests would pass through as there's nothing for CF to cache anyway, but any attack wouldn't even make it to the VPS.
Actually, I know CF offers an SSL thing itself actually, but I'm not sure if it would work with the current bare-bones python server. May be worth a try, all it takes is activating it and changing some DNS records. Worst case scenario, notifications don't work for a few hours... hardly the endof the world, right?

Just my humble two cents!

PS: I still think we could set up a donation pool for server costs, if need be. I'm sure many would be happy to chip in! :)

DonGiotto, 2020-05-14:
@gdombiak I really love your development and I am using it already on all of my printers for now. But I would like to help you out, if possible, with the push service hosting, so that we can make it secure with https and stable enough for some users. I did not find any email, so I just write you here. :-)

MacIsUs, 2021-02-02:
wondering if there is any updates on this? or maybe instructions on how to use my own domain? Im new to octoprint and 3d printers. I'm experimenting with a raspberry pi 2 b that someone gave me ages ago. thanks for any help/advise!
Cheers!

gdombiak, 2021-02-02:
Hey guys,

Thanks for everyones help and input. My concern is not so much about how to set up things but with the hosting cost. As you guys know I try my best to keep this software 100% free. Right now I'm running Lightsail on the cheapest plan which provides an extremely limited environment. My concern is running out of memory or CPU by adding more services to this limited Lightsail instance. This means that maybe by adding this extra workload, I will need to move to the next Lightsail plan.

In short, I'm trying to balance cost versus features and in this case security. Besides the "http is not secure", what would be a real use case of vulnerability that would be a concern and would justify potentially moving to a more expensive Lightsail plan? 

@MacIsUs, unfortunately it is not possible for other people to run this same service on another server. Apple is very strict on security. Any server that wants to send a push notification will need to present the Apple Developer credentials. This means that I would need to give my dev credentials for others to be able to run the server. :(

Regards,
Gaston

MacIsUs, 2021-02-02:
Fair enough.  I’m thinking that maybe you should see how many people are interested in this service.  And get a “beta” test group. Im sure enough of us would chip in enough to cover cost. And once it’s going new users then would have a set subscription fee maybe? Just a thought.  I’m not even sure if this is what I need.  I’d really like to be able to access my Prusa i3 MK3s from anywhere to cancel a print if it begins to fail when I’m not home.  Maybe you have another option to suggest? Or is this even what I’m looking for? Thanks for such a quick reply!!!

callumj, 2021-03-25:
AWS load balancers provide SSL termination - is that something that can bring the cost down?

rickyhoots, 2021-03-30:
> Thanks for everyones help and input. My concern is not so much about how to set up things but with the hosting cost. As you guys know I try my best to keep this software 100% free. Right now I'm running Lightsail on the cheapest plan which provides an extremely limited environment. My concern is running out of memory or CPU by adding more services to this limited Lightsail instance. This means that maybe by adding this extra workload, I will need to move to the next Lightsail plan.
> ... ...
> Regards,
> Gaston

Hi Gaston,

I understand the concern and would propose a solution, albeit workaround from my side. Let me pay for the additional cost. I see the basic Lightsail plan is $3.5 a month, I propose to upgrade to the $10.0 a month (or something similair from another party if your prefer) and I will pick up the tab. It's the least I can do and 'give back' and as an added bonus the app wil be safer and maybe some other new sparkly feature because of the added capacity.

jollino, 2021-03-30:
We should definitely add a prominent option for a tip jar, at least to cover the costs of these services (and I would absolutely not object to Gaston keeping the rest, given all the work he's putting into all of this!)

For what it's worth, interesting alternatives are:
- Hetzner Cloud CX11 (1 vCPU, 2 GB RAM, 20 GB SSD, 20 TB traffic) - €2.96/mo + VAT
- Kimsufi/OVHCloud Starter (1 vCPU, 2 GB RAM, 20 GB SSD, no traffic limit) - €2.99/mo + VAT
- Kimsufi/OVHCloud Value (1 vCPU, 2 GB RAM, 40 GB SSD, no trafic limit) - €4.25/mo + VAT

For reference, the current/basic Lightsail instance has 1 vCPU, 0.5 GB RAM, 20 GB SSD, 1 TB traffic for $3.50/mo (give or take €2.97/mo).

Note: I'm not affiliated with any of these companies, but I've been doing some research for a small project of mine. I did use Hetzner's CX11 VPS for a few months for another test and it worked well for me.

tzahari, 2025-02-08:
Any updates on this?
I would love to see SSL encryption for the push notifications.