gdombiak/OctoPrint-OctoPod #31: 0.2.4 Update loop

barrettj, 2019-12-05:
Did something forget to get tagged properly?  After installing 0.2.4 from the updater it constantly tells me I'm on 0.2.3 and there's an update available.

gdombiak, 2019-12-05:
Hi @barrettj, I pushed the new release and for maybe 5-10 minutes I forgot to update a file that I immediately fixed. I wonder if you updated in that small time window. If you log into OctoPrint, what version of OctoPod plugin do you see installed? Is it 0.2.3 or 0.2.4? 

If you see 0.2.3 then a quick troubleshooting idea would be to uninstall the plugin and install it again. If you changed the plugin settings then take notes of your setup before uninstalling the plugin. Let me know if this helped. I'm at work now so have limited support. :(

Thanks,
Gaston

barrettj, 2019-12-05:
It still shows as 0.2.3 - I think it's because the zip it's downloading (https://github.com/gdombiak/OctoPrint-OctoPod/archive/0.2.4.zip) in setup.py lists the version as 0.2.3 but I really don't know what I'm doing with plugins on this system.

gdombiak, 2019-12-05:
Gotcha. I will update that file then. I thought it was using https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip.

Will let you know once done

gdombiak, 2019-12-05:
Done. https://github.com/gdombiak/OctoPrint-OctoPod/archive/0.2.4.zip now has the proper version number.

Let me know if issue is solved. Thanks for letting me know this.

Gaston

barrettj, 2019-12-05:
Looks like that fixed it - thanks!

closed 2019-12-05 by barrettj