gdombiak/OctoPrint-OctoPod #32: Suggestion: Abort

snowe2010, 2019-12-19:
I've tried creating my own custom command to 'abort' a print, but it continues to print after executing my custom command, even if I set it to stop the motors etc. Any way you could add in an abort ability?

gdombiak, 2019-12-19:
Hi Tyler,

OctoPod already offers support for pausing, resuming, restarting and cancelling your print. You are not the first person that cannot find this "hidden" functionality. I'm out of ideas for how to update the UI to make it clear.

Check out [the FAQ](https://github.com/gdombiak/OctoPod/wiki/FAQ-using-the-app#how-do-i-cancel-resume-restart-or-pause-a-print-job) that covers how to access this functionality. If you have ideas on how to improve the UI, I'm all ears.

Let me know if you were asking something else.

Thanks,
Gaston

jollino, 2019-12-19:
Maybe something like
**Print status ⚙️**
may be more obvious than
**Printed ℹ︎**
?

Or maybe a button that says "Actions" and unfolds to reveal buttons for the actions? (The button and fold would be to prevent accidental hits, kind of like the ℹ︎already acts as a failsafe.)

Just my two cents!

gdombiak, 2019-12-19:
Thanks Daniele for the suggestion. Before the ugly (i) icon I had a gear like you suggested. I will explore the idea of an actions icon. Space is limited for having text in the button. I also added an initial tooltip when you first install the app but not sure if people click quickly on them to make them disappear or if they really achieve the intention.

jollino, 2019-12-19:
Not ugly at all, but the reason I thought about a gear is that it may instinctively give a sense of "there are things you can change" rather than "there are things you can read".
Maybe something like this? (Please excuse my very poor mockup skills... and the language mix!)
[image]
The show/hide thing wouldn't be really necessary (I'm not even sure how hard it would be to add to the section header, would it require subclassing the whole thing?), but may act as an additional protection.

gdombiak, 2019-12-19:
I think that **IS A WINNER**. It's super explicit. 

BTW, I think I used a "tool icon" and not a "gear icon". I can change that too and people will be able to do it from both places?

The other thing people miss is the fact they can set temperature of bed and extruders. Same problem since they do not click on the (i). Any ideas on how to also make it clear that you can set temp for each thing? Maybe using a gear icon will be enough?

Thanks,
Gaston

jollino, 2019-12-19:
"Jollino uses Print Actions. It's super effective!" :D 

What do you mean with "tool icon"? Like the Settings icon below? That would work too but I'm not sure about the size, it may be too small?
For the extruder and bed temperatures maybe you could a "writing pen" icon, so that it suggests that it can be turned into an editable field? I can't find it on https://developer.apple.com/design/human-interface-guidelines/ios/icons-and-images/system-icons/ but I'm thinking about the one that looks like Compose, but having a line at the bottom rather than the whole square.
Or maybe just make it into a read-only text input that becomes editable when tapped, but that would look kind of ugly.

Speaking of temperatures, a little OT: is there any way to change the shortcut temperatures shown in the tooltip?

snowe2010, 2020-01-02:
I'm very sorry for the late reply. I made this Issue before immediately going on vacation. I did not see the initial tooltip on startup, it might have been because I was so flustered with initially getting into 3D printing that I just skipped it, or I might have just forgotten about it. I also do not know how I missed that in the FAQ. I swear I looked in there, but it's the first one so I must have been looking elsewhere. 

I think any of the proposed changes would work. I didn't even realize the `i` was a button until this thread. The gear is much better, but the large bar at the bottom is the best. I find myself frequently cancelling prints as I'm still calibrating for all my filaments and having to open a dialog would be painful over time.

gdombiak, 2020-01-02:
I hear you @snowe2010. I'm playing with multicolor printing with the Palette 2 so having straight access to file name and being able to stop/cancel/etc with one click would be super convenient.

I will go ahead and implement @jollino's recommendation. I like it a lot!

jollino, 2020-01-02:
> I will go ahead and implement @jollino's recommendation. I like it a lot!

I'm a bit of a pest but I sometimes come up with half-decent ideas :P
In all seriousness, thank you for the great app. I'm pretty sure that OctoPod makes a lot of people's lives easier!

gdombiak, 2020-01-02:
Daniele, no pest at all! You help a great deal making OctoPod a better app. I'm also considering replacing the icons with something of newer design. I used what I had at hand that was free but got a lot of feedback that the app looks very dated. :( My software engineer eyes cannot pick up in those things.

gdombiak, 2020-03-19:
Closing this ticket since I think that all feedback has been incorporated into OctoPod. Thanks everyone!

closed 2020-03-19 by gdombiak