gdombiak/OctoPrint-OctoPod #33: Finish notifications send to early

pcjacobse, 2020-02-18:
The "Print complete" notifications are being sent to early, when im printing something where the last layer takes around 10 minutes, at the start of that layer (around 2 minutes in) the push message is send out.

I just noticed it seems to get send when the timelapse completed rendering, that happens after the last Z layer change.

I don't know if it is an issue in this plugin or in the rc of octoprint it self.

Running:
OctoPrint: 1.4.0rc5
OctoPod Plugin: 0.2.4
Prusa MK3S

gdombiak, 2020-02-19:
Hi @pcjacobse,

OctoPrint tracks the print job status and fires events as the status of the print changes. This plugin sends a notification when the [FINISHING event is fired](https://github.com/gdombiak/OctoPrint-OctoPod/blob/4dcd245df511e94c6442239487e8fdf630b5a2f5/octoprint_octopod/job_notifications.py#L66). In my case between finishing and be done it is basically the same. I wonder why on your case it takes minutes. Here you can see [possible states](https://github.com/foosel/OctoPrint/blob/8409fdaa48ff692bbb050086e8d7c09940b58c59/src/octoprint/util/comm.py#L346) that OctoPrint has. It seems to jump from FINISHING to OPERATIONAL. OPERATIONAL is when the printer is idle and ready to print. The logic of this plugin could be smarter to distinguish between OPERATIONAL because you are done printing or OPERATIONAL because you cancelled or OPERATIONAL because or any other flow of events.

jollino, 2020-02-19:
I had a similar issue a while back and I think it ultimately was due to some plugin that for some reason managed to trigger the change early as well (and as @gdombiak explained, it affected anything else that relied on sensing the status change, as I also got the "print finished" message on Telegram).
@pcjacobse Would you mind listing which other plugins you have enabled in OctoPrint, so I can check if I used any of those? I have since cleaned up my installation of OctoPrint so I don't have my own list anymore, but I'm sure I'd recognize them if I saw them.

pcjacobse, 2020-02-19:
Since I'm new to 3d printing i just installed a bunch that youtubers recommended:

Bed Visualizer (0.1.11)
Change Filament Plugin (0.3.0)
Cost Plugin (0.1.3)
Displays estimated cost for each print
Firmware Updater (1.6.1)
Floating Navbar (0.3.1)
Fullscreen Plugin (0.0.4)
Navbar Temperature Plugin (0.11)
OctoPod Plugin (0.2.4)
Prusa ETA override Plugin (0.1.1)
Prusa Mesh Leveling (0.3.0)
Resource Monitor (0.2.0)
Tab Order (0.5.5)
Themeify (1.2.0)
Webcam Tab (0.1.2)

jollino, 2020-02-19:
Um, out of those I use only use these:

> Firmware Updater (1.6.1)
> Fullscreen Plugin (0.0.4)
> Navbar Temperature Plugin (0.11)
> OctoPod Plugin (0.2.4)
> Resource Monitor (0.2.0)

I'm now remembering that I used to have the issue with some layer height plugin (can't recall which one), which for some reason miscounted the layers by one compared to the GCode page and effectively triggered the early "finished" event. Do any of those plugins count layers, by any chance?

gdombiak, 2020-03-19:
Can we close this ticket? Does not seem to be related to this plugin?

Thanks,
Gaston

gdombiak, 2020-04-30:
Closing due to inactivity. We can open a new one if still an issue.

closed 2020-04-30 by gdombiak