gdombiak/OctoPrint-OctoPod #34: [Feature Request]

simpat1zq, 2020-03-01:
Is there any way to have OctoPod send a notification periodically while a print is ongoing? Like send an update pic every X seconds. Maybe even as a premium feature to cover your push notify hosting costs?

gdombiak, 2020-03-02:
HI @simpat1zq ,

Thanks for placing this request. Let me check into this and run some numbers based on current traffic to see what would be the economic impact (if any). I work hard to keep this project free and would like to keep it this way (as long as possible). Been paying the hosting out of my pocket until AWS accepted to sponsor hosting for this year.

Would sending a notification for this milestones work for you: 25%, 50%, 75%, 100%?

gdombiak, 2020-03-02:
Good news. I checked network traffic and increasing frequency of notifications won't imply a jump in $$$. :) At least not from network traffic. I think that CPU and memory should hold fine too. I will make this feature an opt-in just in case.

Here is how network traffic has been growing over time. I think we will be close to 60GB of traffic.

[image]

gdombiak, 2020-03-21:
Hi @simpat1zq, just finished implementing your requested feature. Let me know if you would like to help test it.

Thanks,
Gaston

simpat1zq, 2020-03-21:
Yes. I can help test. Thank you.

gdombiak, 2020-03-21:
Thanks @simpat1zq . You will need to install the development version of the plugin. Use this URL: https://github.com/gdombiak/OctoPrint-OctoPod/archive/dev.zip

Under Settings you will now see the option to choose progress notifications. See
[image]

commit 58dd47cb (2020-03-21): Send notification at different progress percentages

Closes #34
closed 2020-03-25 by gdombiak