gdombiak/OctoPrint-OctoPod #39: Send Test Notification - Sending Failed

csimmons01, 2020-04-14:
I have setup Octopod on my iOS device and can access and control my printer.  When I try to test the notification setting I get a "Sending Failed:" error.

Octoprint.txt log shows:
2020-04-14 14:57:40,743 - octoprint.plugins.octopod - INFO - Notification Response: {"timestamp":"2020-04-14T13:57:40.622+0000","status":400,"error":"Bad Request","message":"Unable to parse form content","path":"/v1/push_printer"}

I've checked my settings and looked through the issues and it seems that I have everything setup properly.  If I try to open http://octopodprint.com in a web browser on my computer I get a White Label Error page.  Is the push notification server down or do I have something setup wrong?

gdombiak, 2020-04-14:
Hi @csimmons01, I just tested it from my computer and I got the notification. The URL http://octopodprint.com/ is correct. The White Label Error page is expected. Are you running the latest version of the plugin? That's probably not the cause for this though. I'm out of ideas for what could be causing this honestly. :( Is the URL to the camera correct? If you copy it to a browser (in incognito) does it show an image?

Gaston

csimmons01, 2020-04-14:
I've still yet to get a test message to work, but I got a push notification message that my bed was up to temperature!  I figured the "White Label Error" was probably expected.

For what it is worth I am running:
OctoPrint Version 1.4.0
OctoPi Version 0.16.0, running on Raspberry Pi 3 Model B Plus Rev 1.3
OctoPod PlugIn version 0.2.5
iOS OctoPod App version 3.2 build 4

The URL to the camera is correct and I can access the camera in OctoPod app.  I have set temperatures and started a print from the iOS app.  In short everything seems to work except the test message.

This is about a 2 hour print, so I'll see if I get push notifications during the print and will let you know.  Thanks for your work on the app and plugin, I think I am really going to like them.  And thanks for the quick response!

Chris

gdombiak, 2020-04-14:
Happy to help Chris. Since you got the notification about bed temp then connectivity is working fine. You can set to get notifications at different % of progress. If you got those notifications then everything it correctly set up (and testing send notification should work too). If notifications for work progress do not work then I would double check that the URL of the camera that you configured in this plugin is correct. However, I would expect that an incorrect URL will end up in a notification without a camera snapshot.

Let me know how it goes.
Thanks,
Gaston

csimmons01, 2020-04-14:
I've done two prints now, and have only gotten the bed temp and print completed alerts, not the 50% alert that is also configured.  None of these notifications have had pictures.  I have numerous instances of the following error in the octoprint.txt log:

2020-04-14 21:37:44,714 - octoprint.plugins.octopod - INFO - Notification Response: {"timestamp":"2020-04-14T20:37:44.672+0000","status":400,"error":"Bad Request","message":"Unable to parse form content","path":"//v1/push_printer"}

My camera URL was using http://localhost:8080/?action=snapshot, but I have switched it to http://192.168.x.xxx:8080/?action=snapshot ("x's substituted for rest of ip).  If I put that address into an incognito window I do get a camera snapshot.

The only other thing I can think of is I have my camera snapshot set to 1640x1232, not the default 640x480.

So I get some notifications (even on my Apple Watch), but not with pictures and not all of the notifications that I have configured.

I'll keep trying things between prints, but I still don't have it 100% figured out.

gdombiak, 2020-04-14:
Thanks for the update. Plugin will send notification about 50% (or whatever percentage you configured) when it gets that exact same percentage from OctoPrint. Funny thing when I was testing this feature is that for prints that move fast OctoPrint skips certain percentages and thus those notifications were skipped. :( I didn't want to overcomplicate the code with logic to detect these cases

Good hint about image resolution. Apple Watch app is probably hitting a timeout trying to download all that image. Could you try lowering the camera resolution and see if notifications go through? Notification should include the camera snapshot and the notification on the watch should also display the image.

csimmons01, 2020-04-15:
I was able to complete some testing.  Removing the following line from my octopi.txt file has fixed my notification issues.  The test message is now successful and I am getting 25/50/75/100% completion messages with images. 

camera_raspi_options="-x 1640 -y 1232 -fps 20" 

The only line left in my octopi.txt that is not commented out is:
camera="raspi"

I tried most of the resolution options for the Raspi cam and none of them would work with the OctoPod notifications, although most worked for webstream and snapshots.  I did not try all resolution and frame rate options, but simply leaving the defaults works.  My web stream and snapshot resolution are not as good, but they are acceptable, especially as I was able to tweak the manual focus on the camera a bit better.

The highest raspicam resolution did not work for streaming or snapshot (camera_raspi_options="-x 3280 -y 2464 -fps 10") in the Octoprint web interface, nor for Octopod notifications.

So in summary OctoPod notifications work with the default Raspberry Pi Camera settings, but none of the other resolution options seem to work, at least for me (even ones that work fine for streaming or snapshots through a browser).

For reference here is a link to how to change mjpeg-streamer parameters https://community.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203

This has more info on Pi Camera modes:
https://picamera.readthedocs.io/en/latest/fov.html#sensor-modes

If I hadn't recently been tweaking my camera, and you hadn't suspected a snapshot issue, I'm not sure I would have figured this out.  Seems like it could easily cause problems for someone in the future.

Thanks for all of your help, at least I have it working, albeit with reduced camera resolution.

gdombiak, 2020-04-19:
Great research and finding @csimmons01. Happy to hear that you got it working with some setup even though it is less optimal.

I have a Logitech C270 and /boot/octopi.txt is configured with 
> camera_usb_options="-r 1280x720 -f 10 -q 75"

In my other setup, I have a raspberry camera v2 and /boot/octopi.txt  has this setup
> camera_raspi_options="-fps 10 -x 1280 -y 720 -quality 75"

Anything else is commented out. Have you tried with bigger resolutions and reducing quality? I found that up to 75% the pixelation is not that bad. 

Let me know if that worked better.
Thanks,
Gaston

csimmons01, 2020-04-20:
I'll try that next time I have a chance.  I tried a bunch of settings, but I never tried the quality options, and I didn't try 10fps with the 1280x720 resolution either.

I'm loving my Octopod app though, even though I had to give up some resolution.  Manually adjusting the focus on the Raspberry Camera v2 made a big difference for me too.

c-hri-s, 2020-04-30:
These seem to be the magic numbers:
camera_raspi_options="-x 1640 -y 1232 -fps 10 -quality 95"

Maximum resolution which still sends sucessfully.

gdombiak, 2020-04-30:
Thanks guys for all the information you shared!!! I now know how to reproduce this error so a fix is coming soon. As it was said, manually reduce the image resolution of the images for now. The fix will try reducing the resolution for push notifications since they don't need to be really that big anyway.

gdombiak, 2020-04-30:
New release of the plugin is out that fixed this issue! Let me know how it works for you. You can now leave image resolution as you wish.

closed 2020-04-30 by gdombiak

csimmons01, 2020-05-05:
I still was not able to use any image settings I wanted.  Only tried a few, but the one suggested by c-hri-s worked for me:
camera_raspi_options="-x 1640 -y 1232 -fps 10 -quality 95"
Other settings gave the "Sending Failed" message when testing.

camera_raspi_options="-x 1640 -y 1232" did not work for me, for example.

gdombiak, 2020-05-05:
Hi @csimmons01, if you are using the new version of the plugin then you can leave the camera at whatever resolution you like and the plugin will automatically will scale down (only if needed) to -x 1640 -y 1232. 

Note that having extreme high resolution will have other impacts and not sure about that high benefits. Streaming of big resolution implies more network bandwidth, CPU and memory usage. These things could potentially impact OctoPrint. From an OctoPod perspective, super high resolution will not make it appear better in a screen size like an iPhone or iPad. And for Apple Watch it might time out and not be able to display your camera on your Apple Watch.

commit b7e4a0b2 (2020-04-30): 1. Use entered (unsaved) snapshot_url for testing notifications and 2. Reduce image resolution if needed

Closes #45
Closes #39