gdombiak/OctoPrint-OctoPod #40: ? memory leak

mselkin, 2020-04-14:
Hi,

I am observing an issue that I have not seen reported previously.
I am running Octopi with the Octopod plugin on a raspberry pi3b+. It might be importune to mention that it is mounted in a 3d printed frame with a touch screen and it autoboots to the chrome browser (so the browser and Octoprint are on the same raspberry pi). I am also using the dashboard plugin which allows me to see the Pi ram usage.

Every time I use my iPhone Octopod app to view my printer progress, the Pi RAM usage creeps up by about 1 - 5% and eventually with enough logins I run out of RAM and crash the raspberry Pi.

Is it possible there is a memory leak?

Any help would be appreciated.

Mark.

gdombiak, 2020-04-14:
HI @mselkin, thanks for reporting this. OctoPod iOS app does not communicate directly with OctoPod for OctoPrint plugin. The expected behavior on the server (i.e. OctoPrint and the RPi) is that when a client connects (e.g. OctoPod) it will have a spike in CPU usage and memory consumption. However, when the client closes the connection the memory should be released (eventually) and things go back to normal.

Based on your description, it seems like memory is not being released (at least not completely). As you open more connections and closes them, I guess that memory keeps slowly going up. If this is what you see happening, then it indeed looks like a memory leak. To troubleshoot this I would start by running the server on safe mode to discard that some plugin is not causing this problem. This plugin should not be the cause since it does not react to client connections. If running in safe mode works fine then reintroduce one plugin at a time to find which plugin causes the memory leak. If the problem still happens in safe mode then your version of OctoPrint might have the memory leak. If you are not running the latest version of OctoPrint then update and retest things. 

Hope this helps. For further troubleshooting or fix you might want to contact OctoPrint's dev or the plugin that caused this issue.

Gaston

mselkin, 2020-04-14:
Hi,

Thanks so much for your advice.
I disabled and then re-enabled each plugin one at a time and was able to determine the culprit for the memory leak is the AutomaticShutdown plugin - I shall contact the author.

Thanks again for your help.

closed 2020-04-14 by mselkin

gdombiak, 2020-04-14:
Nice finding! Thanks for sharing this info.

Gaston