gdombiak/OctoPrint-OctoPod #41: Install error

wacktattoo, 2020-04-17:
Please, what should i do with it?

this code show, on my installation on plugin.

`ERROR: Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/cli/base_command.py", line 153, in _main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/commands/install.py", line 382, in run
resolver.resolve(requirement_set)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 201, in resolve
self._resolve_one(requirement_set, req)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 365, in _resolve_one
abstract_dist = self._get_abstract_dist_for(req_to_install)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 313, in _get_abstract_dist_for
req, self.session, self.finder, self.require_hashes
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/operations/prepare.py", line 194, in prepare_linked_requirement
progress_bar=self.progress_bar
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 465, in unpack_url
progress_bar=progress_bar
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 316, in unpack_http_url
progress_bar)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 551, in _download_http_url
_download_url(resp, link, content_file, hashes, progress_bar)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 253, in _download_url
hashes.check_against_chunks(downloaded_chunks)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/utils/hashes.py", line 80, in check_against_chunks
for chunk in chunks:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 223, in written_chunks
for chunk in chunks:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/utils/ui.py", line 160, in iter
for x in it:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/download.py", line 212, in resp_read
decode_content=False):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 564, in stream
data = self.read(amt=amt, decode_content=decode_content)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 529, in read
raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 439, in _error_catcher
raise ReadTimeoutError(self._pool, None, "Read timed out.")
ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

gdombiak, 2020-04-17:
Hi @wacktattoo, thanks for reporting this. At the of that log entry you can see a read timeout error. That indicates a networking issue. See this issue https://community.octoprint.org/t/plugins-timing-out-during-install/2916/8 that is the exact same problem. 

Try again and see if it was a transient error.

Regards,
Gaston

gdombiak, 2020-04-17:
Another interesting thing to check, this other person was having under-voltage issues and that was affecting network reliability. See https://community.octoprint.org/t/cannot-install-any-plugin-after-fresh-octoprint-install/17752/13

gdombiak, 2020-04-30:
@wacktattoo are you still having this issue or can I close this ticket now?

Thanks,
Gaston

gdombiak, 2020-09-21:
Closing the ticket due to inactivity. We can reopen if this is still an issue.

closed 2020-09-21 by gdombiak