gdombiak/OctoPrint-OctoPod #42: push notifications not working

Linkeb3, 2020-04-19:
http://octopodprint.com/
http://my_pi_IP:8080/?action=snapshot

I've got the "pro" version of the app open on my phone, and send test notification fails everytime. my phone is on the same network as my pi.

gdombiak, 2020-04-19:
Hi @Linkeb3, thanks for reporting this and sharing your setup. Those parameters look fine to me.

What do you mean by the pro version of the app? Do you have OctoPod installed or is this another application that you are running? If you have OctoPod installed then something else to check is what @csimmons01 found in this thread https://github.com/gdombiak/OctoPrint-OctoPod/issues/39. 

Let me know how it goes with that information.
Thanks,
Gaston

Linkeb3, 2020-04-19:
The octopod has a paid version, I did pay to upgrade it. I can see the camera, bed temp, everything in the app. But when I go to send a test message it fails, citing it cannot find an IOS device.

I looked through the other article, that appears to be for bad messages, not no messages.

gdombiak, 2020-04-19:
@Linkeb3 are you maybe using OctoClient? OctoPod is a 100% free and open source client. This plugin only works with OctoPod app (not with OctoClient or any other iOS app)

Regards,
Gaston

Linkeb3, 2020-04-19:
Wow, sorry for the waste of time.. that’s 100% it.

gdombiak, 2020-04-19:
No worries. Give it a try to OctoPod app. Hope you like it enough to use it as your main app. :)

closed 2020-04-19 by gdombiak

Linkeb3, 2020-04-19:
The layout in OctoClient was easier, but the functionality in octopod is superior