gdombiak/OctoPrint-OctoPod #43: multiple printer notifications are all from first printer?

adelyser, 2020-04-26:
Hello, I am running 2 instances of Octoprint on a single Raspberry Pi 4 without any issues. When the print finishes or is 50% complete, the notification in Octopod shows the notification with the title of my first printer, not second printer. In fact, all notifications have the title from the first printer.

Maybe this is an issue with my plugin setup or configuration?

KenLucke, 2020-04-27:
I, too, have this happening, and was about to open a new issue when I saw this one.  I have 4 printers running, all named for the colors of their mods (CR-10v2 Blue, CR-3 Orange, CR-10v2 Yellow, & CR-10v2 Red).  Blue was the first one on which I installed OctoPod.  All the printers are named properly in all preferences (plugins and OctoPrint itself)

I get notifications for all printers, with the appropriate camera shot for the correct printer, but they are all Titled "CR-10v2 (Blue)" - I've been pulling my hair out trying to track this one down.

gdombiak, 2020-04-27:
Thanks guys for reporting this. So notifications for events are correct (including information and camera) but the problem is in the printer name only?

I assume that each OctoPrint instance has its own config.yaml file. Could you share what each printer (2 printers is enough) under plugins -> octopod? 

Thanks,
Gaston

KenLucke, 2020-04-27:
> So notifications for events are correct (including information and camera) but the problem is in the printer name only?

That is correct.

> I assume that each OctoPrint instance has its own config.yaml file. Could you share what each printer (2 printers is enough) has under plugins -> octopod in config.yaml?

They all list "CR-10v2 (Blue) as the printerName.

Seeing that, I tried changing one of them to the correct name in the yaml file, but the notifications were still wrong.  Then I went "DUH!, Eye-Dee-Ten-Tee error!  I only changed the file and did not reload it, so unless it it pulls the config every time, it won't recognize the change..."   

So I restarted OctoPrint, and the notifications from that instance are now correct in title name.  

Wherever OctoPod is getting the name to stuff into its  own configuration in the .yaml file, it's not getting the proper one.

KenLucke, 2020-04-27:
BTW, "Yellow" had several instances of the printerName in it, the first few were wrong, the last few were correct:

```
           printerID: x-coredata://C394080B-363F-48EE-B1E7-CEF190972B19/Printer/p3
            printerName: CR-10v2 (Blue)
        -   apnsToken: C061526726CB91D4B889D2AFFE212BD0EF94F0565087DE509E82335CF3AF8AE1
            date: 2020-03-25 00:31:59.932004
            deviceName: Ken’s iPhone 8 Plus
            languageCode: en
            printerID: x-coredata://6297F7F3-C25D-49D2-A021-1DEBA0294772/Printer/p4
            printerName: CR-10v2 (Blue)
        -   apnsToken: C43F725A45120445841CA73F526376E94B4AEA3016DA85DE09F0C5F85401B6DA
            date: 2020-03-26 06:27:45.448943
            deviceName: iPad
            languageCode: en
            printerID: x-coredata://C394080B-363F-48EE-B1E7-CEF190972B19/Printer/p4
            printerName: CR-10v2 (Yellow)
        -   apnsToken: C061526726CB91D4B889D2AFFE212BD0EF94F0565087DE509E82335CF3AF8AE1
            date: 2020-03-31 05:16:37.795490
            deviceName: Ken’s iPhone 8 Plus
            languageCode: en
            printerID: x-coredata://6297F7F3-C25D-49D2-A021-1DEBA0294772/Printer/p5
            printerName: CR-10v2 (Yellow)

```

KenLucke, 2020-04-27:
BTW, "Yellow" had several instances of the printerName in it, the first few were wrong, the last few were correct:

```
           printerID: x-coredata://C394080B-363F-48EE-B1E7-CEF190972B19/Printer/p3
            printerName: CR-10v2 (Blue)
        -   apnsToken: C061526726CB91D4B889D2AFFE212BD0EF94F0565087DE509E82335CF3AF8AE1
            date: 2020-03-25 00:31:59.932004
            deviceName: Ken’s iPhone 8 Plus
            languageCode: en
            printerID: x-coredata://6297F7F3-C25D-49D2-A021-1DEBA0294772/Printer/p4
            printerName: CR-10v2 (Blue)
        -   apnsToken: C43F725A45120445841CA73F526376E94B4AEA3016DA85DE09F0C5F85401B6DA
            date: 2020-03-26 06:27:45.448943
            deviceName: iPad
            languageCode: en
            printerID: x-coredata://C394080B-363F-48EE-B1E7-CEF190972B19/Printer/p4
            printerName: CR-10v2 (Yellow)
        -   apnsToken: [redacted]
            date: 2020-03-31 05:16:37.795490
            deviceName: Ken’s iPhone 8 Plus
            languageCode: en
            printerID: x-coredata://6297F7F3-C25D-49D2-A021-1DEBA0294772/Printer/p5
            printerName: CR-10v2 (Yellow)
```

They were obviously from each time I changed the API key, so I deleted all but the last one,   made sure the name was correct, saved, restarted OP, and all works fine.  Is there a reason you are saving old API tokens?

Syl20-94, 2020-05-02:
Hello, I had the same problem, and I also had multiple token section which were not correct in name,  then I cleaned the config.yaml file and restart the service and now is solved ... will it be meaningfull to have to possibility to manage these one from plugin configuration page rather than doing some command line editing ?
thank for the tips anyway, it solve something that makes me crazy while I'm controling my printers from my sofa :)

KenLucke, 2020-05-02:
> will it be meaningfull to have to possibility to manage these one from plugin configuration page rather than doing some command line editing ?

It should not have to be "managed" by the user at all.  The plugin itself should either a) clean the file of old API keys and their associated data (why would you keep outdated info around, anyway?), or b) only choose the last data set for its target printer name, which it is obviously doing for the other things stored there.

Have not seen any response from the developer for almost a week since I posted the solution, but I don't know how attentive he is or what other issues he may have going in life, even though there WAS an update to the plugin a couple of days ago (which didn't appear to address this).  This seems as if it should be a simple oversight in which data he's choosing when starting up, and should be easily corrected.

gdombiak, 2020-05-02:
Hey guys,

Based on the information that was shared I see the reason for notifications with wrong printer names. What I'm still not clear is how was the setup done that ended up producing this problem. If anyone can share the steps that ends up producing this error then I will know what to fix.

Being an open source project, I also accept PR with contributed fixes if someone knows what is the root cause and has a fix for it.

Thanks,
Gaston

KenLucke, 2020-05-02:
> Based on the information that was shared I see the reason for notifications with wrong printer names. What I'm still not clear is how was the setup done that ended up producing this problem. If anyone can share the steps that ends up producing this error then I will know what to fix.

I went through the normal setup procedure on one printer (Blue), then cloned my OP setup (via Backup/Restore) to the other (Yellow), then changed the API key for the second printer (Yellow) in OcctoPrint and used my iPad/iPhone to add a new printer and scanned the new API key in the OctoPod app.

gdombiak, 2020-05-02:
Gotcha. I see now how this happened. Backup/restore is meant for recovering the **existing** OctoPrint instance. By using it as a way to spawn **new** OctoPrint instances, the configuration of plugins will be ported and produce the problem described in this issue. I can see how it can be an easy way to spawn new instances though it has this unintended consequences.

I'm now thinking how this could be solved. Of course one way is to not use backup/restore as a way to spawn new instances but since this happened to many of you then I think that we need to do better than saying _"don't do it!"_ :) 

I can think of 2 complementary things:
1. Add a UI that shows all registered devices so you can clean up the ones you no longer want
1. Automatically delete (somehow) devices that have not being used in 2 months or more.

Since the second one will still send notifications with "wrong" printer names for quite some time then I think that option 1 is the way to go.

Other ideas?
Gaston

KenLucke, 2020-05-02:
I can't believe that everyone who experienced this issue did so because of using that method of makiing sure that all plugins and settings are the same on a new machine, so there must be something else going on, but:

I always prefer, as a user, to have control over my own systems, so a list of registered devices with checkbox for active ones and a "delete inactive" option would be where I would fall on preferences.

I'm always a fan of giving the en

KenLucke, 2020-05-02:
> Gotcha. I see now how this happened. Backup/restore is meant for recovering the existing OctoPrint instance. By using it as a way to spawn new OctoPrint instances, the configuration of plugins will [...]

This is true, I know it ports all the preferences, which is why I do it - I don't want to have to install,  one by one, every plugin, and then go into each one and make sure the settings are the same.  

However, once those preferences have been changed in the new instance, those preferences should be the ones that take hold, no different than if you restore from an old backup to the same machine and then reset things you have changed since the backup was made.

gdombiak, 2020-05-05:
Question for @Syl20-94 and @adelyser : Did you guys also use backup/restore for spawning new OctoPrint instances? 

For now my idea is to add a table with the list of registered devices so you can select the ones you want to delete. This could be convenient this case and other cases too. (e.g. security to see who else is getting notifications)

Syl20-94, 2020-05-05:
indeed I used backup, and I understand that the reason why I had this trouble.
I think your idea of adding table will be good enough and will be able to show which instance name is pushed.
will it be complicated to being able to edit the name of one registration but without changing the key, as an extend of the delete?
thank a lot for your time for this new feature

gdombiak, 2020-05-05:
Hey @Syl20-94, thanks for confirming your use case and the solution. :)

Are you interested in being able to edit "deviceName"? I think that I can add that with no side-effects. The property "printerName" is automatically updated from OctoPod when you change the printer name.

Thanks,
Gaston

KenLucke, 2020-05-05:
> Hey @Syl20-94, thanks for confirming your use case and the solution. :)
> 
> Are you interested in being able to edit "deviceName"? I think that I can add that with no side-effects. The property "printerName" is automatically updated from OctoPod when you change the printer name.

Why not give the ability, with the default fill being the printerName.  If changed, a "Revert to Default" button would be appropriate.

I'm always of the preference to give the end user as much flexibility as possible.

Syl20-94, 2020-05-05:
Well, I'm not sure this is really usefull as I've to admit that I don't play renaming device on a regular basis ;)
In my case, I backuped one octoprint and installed on a new raspberry for a new printer and so saved time to setup the octoprint, in this case I deleted initial printer and add the new one, this is why I got these 2 instances in the config.yaml. 
Table will anyway shows it and this is the most important, then in case of long run, if it can save to generate new instance why not ... but it will be for a very limited number of time in the next century ;)
Maybe another way should be to catch printer name and use the instance with this printer name, so we don"t need to have a look onto all of this ? (I tried to figure out if some people have several printer connected on one octoprint ... )

closed 2020-09-21 by gdombiak

taikolito, 2021-02-06:
Thanks for this solution !
I somehow knew deep inside that cloning full octoprint instances through backup wasn’t something clean 😅

taikolito, 2021-02-06:
> Well, I'm not sure this is really usefull as I've to admit that I don't play renaming device on a regular basis ;)
> In my case, I backuped one octoprint and installed on a new raspberry for a new [...]

I’ve read about this several times, it must occur somewhere

commit fb6555ee (2020-09-21): Fix for Python 2 that needs to encode strings for apostrophes and other ascii values

Close #43