gdombiak/OctoPrint-OctoPod #44: OctoPod Notifications Fail To Send (v3.3)

d-lantz, 2020-04-26:
I'm receiving no Phone notifications through OctoPod after I updated to the latest version of OctoPod (v3.3 at the time of this post).  I've changed no other configurations with my setup - running OctoPrint (v1.4) on RPi 4, Ender 3 Pro (using SRK mini-e3 v1.2), and Phone running iOS (v13.4).

Have restarted all devices multiple times, performed test prints, double checked notifications are enabled and that OctoPod is correctly able to connect to my printer.  Additionally, when I click the "Sent test notification" button in the OctoPod Notifications menu in the OctoPrint Settings, I receive a "Sending Failed:" message.

Any insights or assistance would be appreciated, thanks.

vordenken, 2020-04-27:
I get the same message but only if I configure the snapshot url (In my case an external device which works fine on the octoprint snapshot option). Do you have the snapshot url configured?

c-hri-s, 2020-04-30:
Same for me, and can confirm what @vordenken says, if I remove the snapshot URL then the test message sends just fine.

gdombiak, 2020-04-30:
Thanks guys for all the information you shared!!! I now know how to reproduce this error so a fix is coming soon. As it was said, manually reduce the image resolution of the images for now. The fix will try reducing the resolution for push notifications since they don't need to be really that big anyway.

I will close this ticket and use #39 instead for the fix!

Thanks again,
Gaston

closed 2020-04-30 by gdombiak

gdombiak, 2020-04-30:
New release of the plugin is out that fixed this issue! Let me know how it works for you. You can now leave image resolution as you wish.

d-lantz, 2020-04-30:
I've updated to OctoPod Plugin v0.2.6 and can confirm that the test and regular notifications are working now.  Greatly appreciate the quick responses and lightening-fast plugin updates!