gdombiak/OctoPrint-OctoPod #48: 400s for print progress notifications

blalor, 2020-05-24:
I've noticed that I'm only getting notifications for bed temperature threshold, starting from about the 16th of this month.  In `octoprint.log` I'm seeing these errors for failed notifications:

```
2020-05-20 01:31:16,259 - octoprint.plugins.octopod - INFO - Notification Response: b'{"timestamp":"2020-05-20T01:31:16.214+0000","status":400,"error":"Bad Request","message":"Unable to parse form content","path":"//v1/push_printer"}'
```

gdombiak, 2020-05-24:
Hi @blalor,

Are you using latest 0.2.6 version? That same error log was fixed in https://github.com/gdombiak/OctoPrint-OctoPod/issues/39#issuecomment-613677527. Let me know if you are running the latest version and still see this problem.

Thanks,
Gaston

blalor, 2020-05-24:
I am running 0.2.6, I’m afraid.

gdombiak, 2020-05-24:
ok. Would it be possible for you to share/post here the image that http://localhost/motion/100/current returns? I can use it for testing and see what is going on.

Thanks,
Gaston

blalor, 2020-05-24:
Here you go:

[image]

Curl output:

```
$ curl -vvv 'http://localhost/motion/100/current'
*   Trying ::1:80...
* connect to ::1 port 80 failed: Connection refused
*   Trying 127.0.0.1:80...
* Connected to localhost (127.0.0.1) port 80 (#0)
> GET /motion/100/current HTTP/1.1
> Host: localhost
> User-Agent: curl/7.70.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx/1.17.6
< Date: Sun, 24 May 2020 21:42:36 GMT
< Content-Type: image/jpeg;
< Content-Length: 1179648
< Connection: keep-alive
<
```

gdombiak, 2020-05-24:
I think the problem is the file is too big. The resolution is ok but file is too big probably it is using no compression. Would it be possible to increase compression of the jpg file so it is under 1MB?

blalor, 2020-05-24:
I don’t know if I can do that from motion, but I’ll look. That seems like something the plugin should be able to do, tho?

gdombiak, 2020-05-24:
That's a great idea. I can add a setting to control compression of the jpg image. If you are using octopi (and its mjpeg streamer) then follow [this guide](https://community.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203) to control compression quality. You will have to use the -q parameter like in this example:

> camera_usb_options="-r 1280x720 -f 10 -q 75"

or this one

> camera_raspi_options="-fps 10 -x 1280 -y 720 -quality 75"

blalor, 2020-05-25:
I'm using [motion](https://motion-project.github.io) instead of mjpg-streamer so that I can have multiple cameras without needing to juggle a bunch of ports.  Nothing's jumping out at me for reducing image size; I tried setting `picture_quality` to 50 (the default is already 75), but there's no difference in file size.

I think modifying the plugin for the requirements of the APNS service will be a big help!

Robert-Black, 2020-07-06:
I'm having a similar problem - my snapshot image is over 1MB, which triggers an error from http://octopodprint.com

The reason my image is larger than 1MB is because -quality setting for camera_raspi_options in octopi.txt doesn't seems to affect the snapshot's file size. My streaming resolution is -x 1640 -y 1232 because I'm using the new Raspberry Pi High Quality Camera. As soon as it arrived, I upped the resolution (which is fantastic) and suddenly notifications of print completion from Octopod stopped dead  :(

For now, the higher res streaming is more important to me than the notifications, but I miss them! At least this thread has allowed me to understand where it's going wrong. Looking forward to a future version of the plugin that can control compression of the jpg image.