gdombiak/OctoPrint-OctoPod #50: Apple watch face complications do not update unless apple watch app is opened

DolfLund, 2020-05-30:
I love the idea that I can glance down at my watch and see the print process, but it seems the only way to achieve this is to use the icon of the complication (which is almost always wrong) to open the apple watch app, which then refreshes, and then when closed, the face is correct from that refresh, but does not update again.  Even non-silent notifications of 25/50/100 do not trigger the complication to update. 

Is this the current functionality or a bug? Just curious.  Loving your app either way!! thank you

DolfLund, 2020-06-01:
After some (not related) changes, this seems to have fixed itself as of today.

closed 2020-06-01 by DolfLund

gdombiak, 2020-06-01:
Hi @DolfLund ,

Good to hear that you sorted it out. Apple Watch has a small battery so WatchOS places A LOT of restrictions on what apps can do in order to save battery. Part of those restrictions is a very small number of times you can update a complication per day. It also has a budget (limit) for iOS apps to be able to update WatchOS app. And there are other more budgets that make the entire expiration sub-optimal. I guess it is a balance between battery and features but either way Apple Watch users will notice short battery duration and/or suboptimal user experience like you noticed. One more thing, as the battery of your Apple Watch gets discharged, it will also slow down the frequency for apps to run in the background. Instead of running every 15 minutes they may be every 30 mins or not at all.

Because of all these restrictions, I had to come up with some (also suboptimal) ways to refresh complications and still stay within daily budget. For instance, if the progress % has not changed more than 10% then do not update the complication. Or if we exceeded daily budget then try slower communication methods for talking to the WatchOS app.

Anyway, long explanation about a complicated subject but I'm happy to see that things are updating again. Could be because budget got renewed or battery was charged. .... or why not ...maybe some bug somewhere. :)

Take care,
Gaston