gdombiak/OctoPrint-OctoPod #51: Send a custom message with Rest API

fmatray, 2020-06-04:
Hello,
Is is possible to add a rest API to send a custom message to octopod ?
My idea is to have a generic endpoint like '/api/message?api_key=my_wonderfull_key' and POST a command like {message: "Printer is burning", camera_snapshot: true}.  

I'm actually working on a security device and I would like to notify some events.
This feature can be easily reused by someone else for another project.

gdombiak, 2020-06-04:
Nice improvement request. Since this could be a security attack vector, I will expose a setting in the UI and will be disabled by default.

fmatray, 2020-06-04:
Thanks a lot.
To limit security risk, we can also limit this api to some specific users using settings.

I haven't been through all Octoprint documentation but It the idea would be :
- Add a plugin permission "Allow sending message with Octopod API" 
- Create a user with the permission and an API key
- Check the received API-key against the permission

By default the feature should be disabled and NO user should have the permission.

inforsgit, 2020-07-15:
I'm not sure if this would be a new request or an extension to this one, but would it be possible to also send a custom notification via GCODE? This would allow notifications to be sent by script or terminal and would allow the gcode itself to notify of specific interesting points in the print...like: "starting that tricky bridge now" or something.

fmatray, 2020-07-15:
Hello,
Perhaps using a plugin to send the message from M117.

inforsgit, 2020-07-16:
I agree, M117 would be a good start, but maybe some extension to the text, because in a normal print there are lots of M117 messages as the print progresses and it would  be annoying to get them all as notifications.