gdombiak/OctoPrint-OctoPod #55: Remote access

Myrddinwylit, 2020-08-01:
I am getting notifications fine, and the app works great when I am inside the same network as the OctoPrint. Once I leave that network the notifications still work but cannot see or access the OctoPrint. 

The documentation says can connect from anywhere.

greetoz, 2020-08-05:
You will need to do a portforward in your router to the octoprint server (port 80 or 443 if SSL is used).. Or VPN to your home network.

gdombiak, 2020-09-21:
As @greetoz mentioned, you will need to either use _port forward + reverse proxy for secuirity_ or [setup your VPN](https://github.com/gdombiak/OctoPod/wiki/Using-OpenVPN-for-Remote-Access) which sounds harder than it actually is.

closed 2020-09-21 by gdombiak