gdombiak/OctoPrint-OctoPod #59: [Request]

KenLucke, 2020-09-03:
Is there a way,  or can you please add a way, to lock the controls on the iPhone/iPad end?  I have several times accidentally hit one of the Move buttons when moving my device, and ruined perfectly good prints.

gdombiak, 2020-09-03:
Hi @KenLucke,

If you want to lock any operations while OctoPrint is printing then go to Settings -> Security -> Auto-lock while printing and enable that setting. The app will disable all buttons while printing and automatically enable them back once done.

Let me know if this is what you were looking for.
Regards,
Gaston

KenLucke, 2020-09-03:
That will work, thank you!  Somehow, I missed that.

closed 2020-09-03 by KenLucke

gdombiak, 2020-09-03:
Cool! Thanks for reaching out and using OctoPod. Let me know if you have other ideas that want to see implemented.

Gaston

KenLucke, 2020-09-03:
> Cool! Thanks for reaching out and using OctoPod. Let me know if you have other ideas that want to see implemented.

The only thing I'd like to see at this point is a better implementation of the gcode viewer used in OctoPrint - one that can handle larger files.  There's a recent plugin for OP called "Pretty Gcode," but it's not to my liking and a horrible CPU suck, and I don't think it handles much larger files.  

If only there were some way to "farm out" the gcode processing to a separate process outside the browser(s), which is where I believe that the bottleneck/filesize limit lays, and yet keep it tied into the OP interface.

My programming days are long behind me, so it's nothing I can even attempt.