gdombiak/OctoPrint-OctoPod #60: Times do not compare when using Print Genius

mrloofer, 2020-09-06:
So I didn't know if to log this as an issue with OctoPod or Print Genius plugin. So I may do both in the hope that someone knows the answer. When Print Genius plugin is installed, it gives very accurate print estimation times as it uses the estimate in the gcode from Cura. However, the times (and progress) given in OctoPod app are vastly different from what's reported in OctoPrint after the Print Genius plugin is installed. So to me it looks like the Octoprint UI is reporting correctly with the new estimates but the OctoPod plugin is reading the OLD estimates somehow. That's the only reason I can think why they are so off. And when I mean off, I mean at least 20%-30% behind the OctoPrint UI progress bar. The one thing that is exactly the same however is the estimated completion time. That matches perfectly.

gdombiak, 2020-09-06:
Hey @mrloofer,

Thanks for reporting this. OctoPod renders the information that is provided by OctoPrint (see [Retrieve information about the current job](https://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job) and [Push updates (aka Websockets) ](https://docs.octoprint.org/en/master/api/push.html#current-and-history-payload). I do not know how Print Genius plugin is coded to update the information in the web UI. Maybe contact the dev of that plugin?

I use [OctoPrint M73 ETA Override](https://plugins.octoprint.org/plugins/m73etaoverride/) that does the same thing and it correctly modifies OctoPrint to return the accurate ETA through the 2 APIs that I shared above.

Regards,
Gaston

mrloofer, 2020-09-06:
Yeah I didn't know if it this would be more Print Genius related or not. I'll raise an issue with them. Thanks for the tip for the M73 ETA Override plugin, I'll give that a try and see if the results are better. Cheers.

closed 2020-09-06 by mrloofer