gdombiak/OctoPrint-OctoPod #63: Webcam URL incorrect in ios app

dayfaerer, 2020-09-22:
I just downloaded this to my pi, plus the iOS app, and the webcam streaming url isn't correct. It has 2 / marks where there is only supposed to be one / mark. Ive fiddled around on the app and couldnt find a setting to change it, and i have no knowledge of coding so i dont understand anything on the pi, and was wondering if there was a simple fix for this

gdombiak, 2020-09-22:
Hi @dayfaerer,

Could you go to Settings -> Printers in OctoPod and verify that the URL does not end with / ? If it does could you remove it and try again? Let me know if that was the issue and I will fix this for the next release.

Thanks,
Gaston

gdombiak, 2020-09-22:
Moving ticket to proper repository.