gdombiak/OctoPrint-OctoPod #69: Very high data usage when over cellular

BlaydeRunner08, 2020-11-01:
I recently noticed that I'm noticing that OctoPod seems to take an incredible amount of cellular data.

Typically, I used OctoPod to keep an eye on my print while I'm on the same wifi, but occasionally I use it to monitor it remotely.  Usually, when I'm out, I'll open the app, check the print/webcam, and then turn my phone off.  However, I just went to check my data usage, and I noticed that OctoPod used up 2GB in the last month!!!

Does the app continue to run the webcam stream even when it's running in the background?  Has anyone else experienced this? 

[image]

gdombiak, 2020-11-01:
Hi @BlaydeRunner08,

Couple of comments with the hope we can bet to the bottom of this:
* Cellular data usage in that screen is not monthly but of the period. Period means since last reset which you can do at the bottom of that page
* iOS does not let apps run constantly in the background so no constant data usage in background is causing this
* Are you using iOS 14 widget? If you do and you are using the widget with an Image then remember that this image will be downloaded/refreshed every 5 minutes when you are printing. So if you are printing all day long AND you add the new iOS 14 widget that has a camera then it will be refreshed every 5 minutes. 
* If you added several widgets with camera and all those printers are printing then all those images will be downloaded every 5 minutes
* For big data usage I would look at the size of the image that is being downloaded. The better resolution then the more data it will use. 
* Big image size combined with refreshing 1 or many iOS 14 widgets that have an image will consume lots of data
* Another option, big image size and no iOS 14 widgets ... then consider that OctoPi uses an inefficient streaming method. It uses MJPEG which basically means that it will send a sequence of JPEGs images to give the illusion of a movie. If you have a high FPS (frames per second) then it will send a big number of images per second. And if your images are high resolution then each image will consume a big amount of data. If you open the app and watch the camera for a minute or more then a big big data usage will happen

In summary, things to check for saving data usage:
* Before doing anything, verify period in which the 2 GB of data usage was downloaded. If period is of several months then this might be acceptable. You might want to reset the period/statistic and monitor consumption when not at home. 
* Use smallest iOS 14 widget that does not fetch camera. No image = way less data usage
* Reduce camera resolution. High resolution = big images = big data usage
* Reduce FPS of camera. 

Hope any of this helps find out what is going on
Let me know if you have further questions.
Gaston

BlaydeRunner08, 2020-11-02:
Hey @gdombiak, I'll try and answer all the comments you made here, as concisely as I can!

- I do understand that the data usage is cumulative, however, I reset my stats a month before for other reasons, so this usage is only for the last month.
- I did have a iOS widget on a secondary screen, but it was a small widget with no camera fetch.  I have since removed that widget.
- My current camera settings are 1080p @ 15FPS, but I notice that I will get more like 1-2 FPS when away from the house over data.

Going forward, what should I try to reduce data usage, other than reducing quality?  You say that the iOS doesn't allow apps to run constantly in the background, but is there a time limit that they are?  If I disabled background app refresh, would I see any decrease in data usage?

BlaydeRunner08, 2020-11-16:
@gdombiak, any insight to this after my additional information?

gdombiak, 2020-11-16:
Hi @BlaydeRunner08,

Sorry for the late reply. iOS will let an app run a few seconds in the background. I would try doing:
1. Reset stats
1. Disable background refresh

Give that a try and see if that made any difference. If not then try:
1. Reset stats
1. Lower camera resolution

Let me know which combination solve this issue.

Regards,
Gaston