gdombiak/OctoPrint-OctoPod #72: paused_for_user notification not working

pheuts, 2020-12-20:
I'm having trouble getting the paused_for_user notification to work. 

Al other notifications work just fine, so I know the connection to the iOS app is as it should. I set the notification delay for this type of notifications to 1 minute. 

I'm using the Smart Filament Sensor Plugin to trigger an M600 command, and added the M600 as a pausing command to OctoPrint. The printer reacts as it should, and OctoPrint shows the printer being paused and gives the possibility to "resume".

Even after a few minutes, stil no notification is send to the iOS app.

The OctoPrint log also does not show the event.

I'm I forgetting to configure something? Or is the OctoPod plugin somehow not able to catch the M600 command from another plugin?

gdombiak, 2020-12-27:
Hi @pheuts, could you post what you see in the Terminal of OctoPrint? OctoPod plugin is looking for a line that starts with **echo:busy: paused for user**. Marlin will print that line on the terminal when it pauses the printer (for whatever reason). If you see that line then check your configuration for _Printer Paused Notifications_. Make sure that you do not have a value of 0 since that will disable the pause notification. If progress is more than 0% and less than 100% then you should see a notification.

gdombiak, 2020-12-27:
I created this sample gcode that does not print but moves the printer and executes an M600. I printed that file from OctoPrint and I got the notification as expected. I know it is now your use case but wonder if that works for you. It will help narrow down the use case.

[pause.gcode.zip](https://github.com/gdombiak/OctoPrint-OctoPod/files/5744114/pause.gcode.zip)

pheuts, 2020-12-27:
Hi @gdombiak,

Thank you for your help. It is much appreciated.

Your first message explained to me that the plugin is triggered by an response from the printer, not a command send to the printer. And that's where the problem lies. My printer's (AnyCubic i3 Mega-S) firmware does not support M600 commands. I added M600 to OctoPrints pausing commands, and use OctoPrints Gcode pausing scripts to make the printer do what i need for a filament change. But this does not make the printer give any response, so no notification from OctoPod is triggered.

I tried triggering a notification in a bit hacky way, by sending "echo:busy: paused for user" from my pausing scripts. Which gives the following in the terminal:
Send: echo:busy: paused for user
Recv: echo:Unknown command: "echo:busy: paused for user"
I hoped you used a contains-conditional. But no luck. I assume you use .startswith() ?

Any chance you are willing to make a change somewhere to make this work? Either change the conditional to something like:
if "echo:busy: paused for user" not in [string]: 
    continue

Or also check commands send to the printer? Or anything else you think is best :)  (I'm just a noob at programming)

Thanks in advance!

gdombiak, 2020-12-29:
Hi @pheuts, does your printer's firmware support M0 gcode? Marlin implements it, it will stop the printer and send back to OctoPrint the "echo:busy: paused for user" message. I wonder if this other firmware does that as well. We can explore other ideas as you suggested if not. BTW, your logic is very good so not a noob on my book. :)

pheuts, 2020-12-30:
@gdombiak, as you suggested I've been trying with the M0 (and M1) command. The printer does react to the command, but not the way anybody would want it to. It stops printing, but leaves the hotend on the object, gives no serial respons what-so-ever, and times out on serial. I'm gonna leave those commands in the blocked list from OctoPrint. I guess they are there by default for a reason.

Looking through Marlin's gcode's I've also tried M6, M25, M118 and M226. But al no luck.

Any changes on your side would be much appreciated.

andrhamm, 2021-02-27:
A relatively new feature in the PrusaSlicer UI is to add a pause at a specific layer. It would be nice if these kinds of pauses would result in the pause notification (currently the notifications aren't firing for pauses configured in this way). Better yet, it would be handy to be able to use the similar PrusaSlicer feature where you can do the same thing but for any custom G-Code... it would be nice to be able to insert a special g-code that OctoPod could see and send the notification. For example, I would actually like to be notified a few layers _before_ the pause layer, so I can get to my printer and be ready to quickly, for example, insert magnets into the print.

gdombiak, 2021-02-27:
Hi @andrhamm, 

Thanks for the detailed ask. Quick answer before I think deeper into your request. :) If you are using the [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) plugin then you can set a notification from OctoPod for any layer you want. It is not the same as you requested but was wondering if that will help you until I implement your ask.

Regards,
Gaston

andrhamm, 2021-02-28:
I do have that installed but I was not aware of that feature, I will
check it out, thanks!

—
<https://github.com/gdombiak/OctoPrint-OctoPod/issues/72#issuecomment-

auth/AAEQNMLQ3DHGZFTAADMQXILTBFBHRANCNFSM4VDMCB2Q>.

kwalkerk, 2022-05-07:
I have Octopod running on an iPhone.  I don't see a "main bell" in the Octopod app.  I'm trying to get notifications for a pause at layer to change filament.  I do have the DisplayLayerProgress plugin installed

gdombiak, 2022-05-08:
Hi @kwalkerk,

Have you configured your slicer to include layer information? OctoPod displays layer information and lets you receive notifications on layers when DisplayLayerProgress plugin installed and gcode has layer information which requires slicer configuration. If OctoPrint web ui displays layer information then something else must be going on.

Let me know how it goes.
Thanks,
Gaston

kwalkerk, 2022-05-08:
Thanks for the reply.

As usual it was my misunderstanding of how things worked.  I had everything set up properly, but was not using the “bell” on the iOS app properly.

I use the “Pause at Height” plugin.  This plugin must not insert the proper codes into the code, because it does not trigger any notification from OctoPod. I have the ability to add gcode before the layer.  Is there something I can add which will trigger the notification?  What is the trigger?

BTW, thanks for this plug in  I love it

gdombiak, 2022-05-08:
Gotcha. Yes, these are different functionalities. Describing each one in case any of them help you.

### OctoPod + DisplayLayerProgress plugins: Notification when print reaches a certain layer
When both plugins are installed and your slicer is correctly configured according to DisplayLayerProgress plugin, then any newly generated gcode will include layer information. From OctoPod iOS app you can use the bell button next to layer information to receive a notification (with an image) when print job reaches the requested layers

### OctoPod plugin: Notification when print is paused waiting for user
If you are using Marlin firmware and your gcode includes M600 or runs out of filament and you use a filament sensor then OctoPod plugin will send you a notification indicating when printer halted and is waiting for user. This is also great for swapping filaments. 
If you are using Prusa Mini then this functionality only works when printing from OctoPrint. The firmware of prusa mini does not send proper notifications that OctoPrint can react to.

### “Pause at Height” plugin: Notification with custom gcode
I do not use Cura so I never heard of “Pause at Height” plugin. I read its documentation and it seems like you can add whatever gcode at any layer. If you are going to include M600 then what I described in the previous section still applies. If you are including any other gcode then OctoPod plugin will not send any notification since it is "unknown" gcode. 

Let me know if you are injecting other gcode and I may improve OctoPod plugin so that you can receive notifications on custom gcode. This was requested once before but there was not much demand for it so it is low in the priority list.

Regards,
Gaston

kwalkerk, 2022-05-08:
Thanks so much for the detailed explanation!

The M600 did the trick.  The plugin allows you to send Code before the pause.  I inserted the M600 and got the behavior I wanted.

Thanks,

Ken