gdombiak/OctoPrint-OctoPod #74: Push notification thumbnail for iPhone upside down

jghake, 2021-01-16:
I have two cameras set up, a raspberry pi camera on port 8080, and an emeet webcam on 8081. I've set the snapshot URL properly for the emeet but the thumbnails on my phone are upside down while the camera view is not in OctoPrint.
[image]

gdombiak, 2021-01-17:
Thanks @jghake for sharing this. I think I know what is happening. I do not think I implemented image rotation. Since you can use the URL of any camera .... I will add new buttons to rotate the image in the plugin configuration. I cannot rely on OctoPrint's configuration for this.

Thanks,
Gaston

gdombiak, 2021-01-17:
@jghake are you using OctoPrint's camera with this plugin? I checked the code and I see that the plugin will rotate/flip the image according to the settings you have in here:

[image]

Is that not working for you? I will still do this improvement since I think the flip/rotate settings should be independent since you can set any URL for the notification thumbnail.

Gaston

jghake, 2021-01-17:
Here is my camera configuration. I'm using the secondary emeet 960 as the notification URL.

[image]

[image]

freilich, 2021-01-17:
The "flip" flags in OctoPrint only affect the OctoPrint UI.  If you go to the raw stream (/webcam/?action=stream) it will be the natural orientation of the camera, which is likely why the thumbnail is not correct.

jghake, 2021-01-18:
The thing is it isn't upside down.

[image]

freilich, 2021-01-18:
I think I see the issue.  You have two cameras, the first one needs to be flipped but the second one doesn't, right?   Looks like the code doesn't differentiate between them, it just sees that you have the checkboxes marked to flip so it flips the thumbnail.  Editing /boot/octopi.txt to flip the camera, then unchecking the boxes in the UI should solve the issue for you.

jghake, 2021-01-23:
Tried this and all is well.

gdombiak, 2021-01-24:
Added new configuration so you can control flip/rotation for whatever camera you are using.

[image]

closed 2021-02-07 by gdombiak

gdombiak, 2021-02-07:
FYI, new release is out that now let's you flip/rotate cameras from the plugin settings page

commit c5cf7cec (2021-01-24): Added configuration to rotate cameras. Initial migration will copy existing values from OctoPrint's camera

Close #74