gdombiak/OctoPrint-OctoPod #76: Print percentage issues

jghake, 2021-01-24:
I've got M73 plugins, and the PrintTimeGenius plugin installed for much more accurate print times. Octopod is sending me alerts that don't align with what is showing on the printer screen due to these plugins.

[image]
[image]

akiltopu, 2021-02-04:
same problem :(

gdombiak, 2021-02-04:
Hey guys,

I do not use PrintTimeGenius plugin so asking this. Does it only change remaining print time or does it also change percentage of print progress? M73 plugin only changes remaining time (to be more accurate) but it does not change progress %.

Thanks,
Gaston

akiltopu, 2021-02-04:
it changes only progress %. Remaining time is accurate

jghake, 2021-02-08:
Sorry it's been busy lately. Print time plugin runs the gcode through a simulation to calculate the print time more accurately. It then updates the estimate for the file, and handles the % updates throughout the print. M73 plugin, sends these updates to the printer so it displays on the screen (tft35 in my case).

tverett, 2021-02-16:
I'm not using the M73 plugin but I am running the printtimegenius as well as ProgressBasedOntime plugins and my progress bar as well as time remaining is correct in both the web octoprint GUI as well as in octopod app (they always match each other) however the % alert/notifications I get from Octopod are not in sync with those progress bars.

jimwhitelaw, 2022-05-10:
I noticed something like this recently too. In the example screenshot below, I'd just got a 50% notification, but the print is actually closer to 75% done based on the times shown. (~2hrs remaining of ~7.5hrs)
[image]