gdombiak/OctoPrint-OctoPod #77: 5% or 10% notification option to allow for early catch of failed prints

Weirdosrock, 2021-01-27:
Can we please have added a 5 or 10% complete notification to allow for the catching of adhesion failures in a given print?

gdombiak, 2021-01-27:
Hi @Weirdosrock,

What I do to check for early failures at different points of the print is the following:
1. Install [DisplayLayerProgress plugin](https://plugins.octoprint.org/plugins/DisplayLayerProgress/)
2. Start print
3. In OctoPod, click on the notification bell icon next to layer
4. Set list of layers you want to be notified at. When print reaches specified layers, you get a notification

With this trick, if I have a super long layer 1 then I set an alert at layer 2, for some risky support-less areas I also use this trick. 

Would this help?
Gaston

andyylin, 2021-08-24:
Would it be possible to create this as an option, for all prints?

Even better if OctoPod could be set to show notification with picture for first X layers.

gdombiak, 2021-08-24:
@andyylin I like a lot the idea of sending a pick for the first X (up to 5) layers!!! Will add this to next release of plugin. Thanks for sharing this great idea.

Gaston