gdombiak/OctoPrint-OctoPod #80: No snapshot at 50% Version 0.3.0

Jack77777777, 2021-02-17:
I have configured to get notifications at 50 and 100%. Since 0.3.0 50% includes no snapshot. 100% does

Jack77777777, 2021-02-23:
@gdombiak is this a bug or a feature ?

gdombiak, 2021-02-24:
Hi @Jack77777777 , not a feature for sure. :) I made some changes in that area of the code in the last release so will check out what is going on. Thanks for letting me know.

Gaston

Jack77777777, 2021-06-02:
@gdombiak Do you have a release date for 0.3.1 ?

gdombiak, 2021-08-29:
Hey @Jack77777777, just to confirm, you are getting the notification at 50% but it does not include a camera snapshot? Is this happening always? Sounds similar to https://github.com/gdombiak/OctoPrint-OctoPod/issues/118

gdombiak, 2021-08-29:
I know what is going on.... I will include the fix for next release. I'm sure there was an error in the logs

commit a9454c5a (2021-08-29): 1. Fix exception that prevented progress notifications from including a camera snapshot and 2. Respect rotation settings when sending camera snapshot for progress notifications

Closes #119
Closes #118
Closes #80
closed 2021-08-29 by gdombiak