gdombiak/OctoPrint-OctoPod #81: Suppress a notification

ingo916, 2021-02-21:
I use TP-Link smart outlet to turn off power to my printer but octoprint is on a different outlet and never shuts off. However octoprint detects the serial connection as down and generates a notification. After installing octopod I get many many sms messages

is there a way to suppress theses alerts/push notifications from octopod?
 
[image]

tibmeister, 2021-02-26:
Yes, I am having this issue to where it's spamming my phone and watch, about 2 per minute.  This is new behavior and will annoy me to no end.

ingo916, 2021-02-26:
I resolved the issue by removing the tape I added to the +5v on the usb plug that attaches to my printer and raspberry pi.

taclane, 2021-03-25:
The error is generated by Octoprint itself when it fails to automatically connect to a printer, usually after a reboot.  I'm not sure if there's an easy way for the Octopod plugin to separate out this particular error from others it receives from Octoprint.

If you are using the TP Link plugin, an easy way to stop this notification it is to disable the Octoprint auto-connect, and let the plugin handle connect/disconnect.
[image]

[image]

Alternatively, if you want to leave Octoprint auto-connect enabled, you can do a quick edit to `/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py` around line 3265 to comment out the error, and flag the printer as offline instead.
```python
#        error_text = "No more candidates to test, and no working port/baudrate combination detected."
#        self._trigger_error(error_text, "autodetect")
        self._changeState(self.STATE_CLOSED)
```
(whitespace is important, there are 8 spaces in front of self._changeState)

closed 2021-08-28 by gdombiak