gdombiak/OctoPrint-OctoPod #89: Feature Request: Receive notifications from 3rd part plugins

SinisterRj, 2021-06-11:
Hello,

I`m starting to develop a new OctoPrint Plugin that's monitors and shuts down the printer for safety reasons ([Safety Printer](https://github.com/SinisterRj/SafetyPrinter)) and it would be nice if I can use your plugin to send notifications when something wrong happens. It's possible?

gdombiak, 2021-07-03:
Hi @SinisterRj,

This is a great idea. I will see how we can open the door so that other plugins can send push notifications to OctoPod app with custom messages. Would be good to support all translations but we can start simple for now. Filing for next release.

Thanks,
Gaston

gdombiak, 2021-07-06:
Good news @SinisterRj. I implemented this feature which will be released in the next version. I also wrote this [wiki page](https://github.com/gdombiak/OctoPrint-OctoPod/wiki/How-to-send-push-notifications-from-3rd-party-plugins%3F) that shows how to use this functionality

closed 2021-07-06 by gdombiak

SinisterRj, 2021-07-29:
Very nice Gaston! It's already working, Thanks!
In your wiki I recommend to substitute:
`self.push_notification("My message goes here")`
by:
`try:
    self.push_notification("My message goes here")
 except AttributeError:
    pass`

to avoid AttributeError if the user installs the 3rd part plug-in but not Octopod.

gdombiak, 2021-07-29:
Great catch. Updated wiki with your good suggestion. And happy to hear it is working fine for you!

Gaston

commit e87683c5 (2021-07-06): Added support for 3rd party plugins to send push notifications

Closes #89