gdombiak/OctoPrint-OctoPod #94: Unable to enable plugin - Invalid Syntax error in logs

BlaydeRunner08, 2021-07-07:
I recently noticed that I wasn't receiving any notifications from OctoPod on print progress, so I went to investigate, and I noticed that the plugin was disabled.  I attempted to enable it, but after a restart of OctoPrint, it always disables again.

I uninstalled and cleaned up data and re-installed, but the same issue every time.

Checking logs, I see this error pertaining to OctoPod:

`
2021-07-07 08:04:25,036 - octoprint.plugin.core - ERROR - Invalid syntax in /home/administrator/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octopod/__init__.py for plugin octopod
Traceback (most recent call last):
  File "/home/administrator/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 711, in _parse_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python2.7/ast.py", line 37, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "/home/administrator/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octopod/__init__.py", line 360
    def push_notification(self, message: str, image: Image=None) -> bool:
                                       ^
SyntaxError: invalid syntax
`

Any idea what could have caused this all of a sudden? Only other change I've made is update a couple plugins...

a7hybnj2, 2021-07-08:
I will have to check my logs to make sure its the same error but I too cannot load this plugin. I got a notification today that there was an update available and I updated. After that it fails to load.

BlaydeRunner08, 2021-07-08:
> I will have to check my logs to make sure its the same error but I too cannot load this plugin. I got a notification today that there was an update available and I updated. After that it fails to load.

@a7hybnj2 what OS are you running OctoPrint on?  I'm on a standard computer running Ubuntu.

a7hybnj2, 2021-07-08:
> @a7hybnj2 what OS are you running OctoPrint on?
OctoPi on a raspberry pi 3(I think)

gdombiak, 2021-07-08:
Hmm, so this error started to happen after you guys installed the update? If that’s the case, could you tell me if you’re using python 2 or python 3? 

Sorry for the inconvenience. 
gaston

BlaydeRunner08, 2021-07-08:
I have both installed, but according to the error log, it looks like it's running on 2.7

`python -V - Python 2.7.17`
`python3 -V - Python 3.6.9`

gdombiak, 2021-07-08:
Could you switch to python 3 for a quick test? Almost sure this is the root cause. I didn’t test the new version with python 2. 🥲

BlaydeRunner08, 2021-07-08:
I still have 5 hours on a print, but I’ll give it a go tomorrow!

gdombiak, 2021-07-08:
Thanks @BlaydeRunner08. Tomorrow I’ll fix this and release a new version. I think I know what broke it.

BlaydeRunner08, 2021-07-08:
@gdombiak updated my OctoPrint to Python v3 using https://octoprint.org/blog/2020/09/10/upgrade-to-py3/ and can confirm that OctoPod is running normally again!

a7hybnj2, 2021-07-08:
I have another 12 hours on my print before I can mess with it.

a7hybnj2, 2021-07-08:
This should prob be a diff issue but is applicable to this conversation. OctoPrint-OctoPod is listed as >=2.7, <3...which technically means doesn't support py3.

BlaydeRunner08, 2021-07-09:
@a7hybnj2 mine shows as “Python: >=2.7,<4“

a7hybnj2, 2021-07-09:
[image]
¯\\_(ツ)_/¯

gdombiak, 2021-07-09:
Good news! I recreated my Python 2 environment and fixed the incompatibility. Tomorrow will push new version that solves this issue.

Sorry for the inconvenience.
Gaston

BlaydeRunner08, 2021-07-09:
@gdombiak well now that I’m up to python3, I no longer need the fix, but I’m glad I was able to help troubleshoot this for the rest of the v2 users!

closed 2021-07-09 by BlaydeRunner08

gdombiak, 2021-07-09:
@BlaydeRunner08, updating to Python3 was a good thing for you. Python 2 has been abandoned and only Python 3 is being maintained now. I will still fix this issue for those people that cannot update to Python 3 or do not know how to upgrade their systems.

commit 47ba5d82 (2021-07-09): Fixed incompatibility with Python2

Closes #94