gdombiak/OctoPrint-OctoPod #97: Migration error thrown on fresh install with latest OctoPi install.

Pizzahd88, 2021-08-03:
2021-08-03 20:52:59,767 - octoprint - ERROR - Error while trying to migrate settings for plugin octopod, ignoring it
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 512, in init_settings_plugin_config_migration_and_cleanup
    implementation._identifier, implementation
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 496, in settings_plugin_config_migration_and_cleanup
    settings_migrator(settings_version, stored_version)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octopod/__init__.py", line 129, in on_settings_migrate
    if current <= 2:
TypeError: '<=' not supported between instances of 'NoneType' and 'int'

Pizzahd88, 2021-08-03:
Afterwards I cannot connect my IOS devices, nofitifications does not work.
[image]

Pizzahd88, 2021-08-03:
More info on my boot:
2021-08-03 20:52:56,243 - octoprint.startup - INFO - ******************************************************************************
2021-08-03 20:52:56,246 - octoprint.startup - INFO - Starting OctoPrint 1.6.1
2021-08-03 20:52:56,246 - octoprint.startup - INFO - ******************************************************************************
2021-08-03 20:52:56,507 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-08-03 20:52:56,550 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-08-03 20:52:56,551 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2021-08-03 20:52:56,551 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-08-03 20:52:56,561 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2021-08-03 20:52:56,594 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-08-03 20:52:57,785 - octoprint.startup - INFO - Added settings overlay from plugin dashboard
2021-08-03 20:52:57,803 - octoprint.plugin.core - INFO - Found 29 plugin(s) providing 29 mixin implementations, 71 hook handlers
2021-08-03 20:52:57,972 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-08-03 20:52:57,979 - octoprint.server - INFO - Intermediary server started
2021-08-03 20:52:57,995 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2021-08-03 20:52:58,241 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2021-08-03 20:52:58,244 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2021-08-03 20:52:58,245 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2021-08-03 20:52:58,245 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2021-08-03 20:52:58,245 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2021-08-03 20:52:58,245 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2021-08-03 20:52:58,246 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2021-08-03 20:52:58,246 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2021-08-03 20:52:58,246 - octoprint.server - INFO - Added new permission from plugin dashboard: PLUGIN_DASHBOARD_ADMIN (needs: "Need(method='role', value='plugin_dashboard_admin')")
2021-08-03 20:52:58,246 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2021-08-03 20:52:58,247 - octoprint.server - INFO - Added new permission from plugin klipper: PLUGIN_KLIPPER_CONFIG (needs: "Need(method='role', value='plugin_klipper_admin')")
2021-08-03 20:52:58,247 - octoprint.server - INFO - Added new permission from plugin klipper: PLUGIN_KLIPPER_MACRO (needs: "Need(method='role', value='plugin_klipper_admin')")
2021-08-03 20:52:58,247 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2021-08-03 20:52:58,247 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2021-08-03 20:52:58,248 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2021-08-03 20:52:58,248 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2021-08-03 20:52:58,249 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
[... cut ...]
2021-08-03 20:53:05,400 - octoprint.plugins.dashboard - INFO - Dashboard started
2021-08-03 20:53:05,422 - octoprint.plugins.octopod - INFO - OctoPod loaded!
2021-08-03 20:53:05,466 - octoprint.plugins.themeify - INFO - Themeify initialized.
2021-08-03 20:53:06,051 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2021-08-03 20:53:06,349 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://octopi.localdomain/', 'path': '/', 'query_string': 'l10n=en'}

gdombiak, 2021-08-28:
Hey @Pizzahd88 , sorry I missed this ticket. Are you still seeing this error?

commit b0a7b011 (2021-08-28): Python 3 will throw an error when current is None and is compared to a number

Closes #97
closed 2021-08-28 by gdombiak