gdombiak/OctoPrint-OctoPod #106: Add 'advanced' thermal runaway detection

gdombiak, 2021-08-14:
Detect when hotend is not warming up and remains in same temperature. This can cause a fire since heater may continue to heat up while thermistor is not reading increase in temp. Could be because of some hardware failure. Printers whose firmware does not detect thermal runaway will not detect this case so this alert may save someone from starting a fire!!!

tideline3d, 2021-08-18:
Here's a log with false alarms happening.  
[Explore-logs-2021-08-17 21_29_30.txt](https://github.com/gdombiak/OctoPrint-OctoPod/files/7003883/Explore-logs-2021-08-17.21_29_30.txt)

tideline3d, 2021-08-18:
Getting better with the latest dev build, but still a number of false warnings today.  You can see the # of instances of the error have gone down since yesterday.  
[image]

Here's a log from today (8-18-20).  

[Explore-logs-2021-08-18 17_18_14.txt](https://github.com/gdombiak/OctoPrint-OctoPod/files/7010241/Explore-logs-2021-08-18.17_18_14.txt)

gdombiak, 2021-08-18:
Yes, it is getting better. In the logs I see 2 instances of alerts and both are because of the same reason: "After 5 seconds temp should have gone up and it went down".

>2021-08-18T14:30:48-04:00	2021-08-18 18:30:48,574 - octoprint.plugins.octopod - WARNING - Possible thermal runaway detected for bed. Actual 67.1 and Target 85.0 
2021-08-18T14:30:43-04:00	2021-08-18 18:30:43,567 - octoprint.plugins.octopod - DEBUG - Thermal runaway - Started to track bed. Temp should go up. Actual 67.5 and Target 85.0

>2021-08-18T13:52:14-04:00	2021-08-18 17:52:14,887 - octoprint.plugins.octopod - WARNING - Possible thermal runaway detected for bed. Actual 84.8 and Target 90.0 
2021-08-18T13:52:09-04:00	2021-08-18 17:52:09,879 - octoprint.plugins.octopod - DEBUG - Thermal runaway - Started to track bed. Temp should go up. Actual 84.9 and Target 90.0

I will handle this case so that if after 9 seconds the temp did not go up then send the alert. Sounds good?

Thanks,
Gaston

tideline3d, 2021-08-18:
Yep, or you could put a fudge factor in since its only off by less than 1*c.  Give it a little buffer to bounce around a bit.   I wasn't doing anything special with them today, most of these were either during initial preheat (from the LCD menu) or shortly after starting the print.

tideline3d, 2021-08-19:
One error today, this time with the nozzle temp.  I had the printer set to pre-heat and it had just reached 230 and was overshooting a bit.  I started the print and looks like the hot end was cooling down a bit right when I clicked start and it set the temp to 240.  

[image]

gdombiak, 2021-08-19:
Interesting. Temp went down more than 1C when warming up. I guess I was being too aggressive/optimistic. Update again from dev, I now allow up to 2C margin. Hope we got them all now! :D

commit 8279ca77 (2021-08-14): 1. Added 'advanced' thermal runaway detection and 2. Improved another false alert of thermal runaway

Closes #105
Closes #106
closed 2021-08-20 by gdombiak