gdombiak/OctoPrint-OctoPod #122: iOS App Notification shows "Unknown code"

paul-tian, 2021-08-31:
Recent OctoPod update on OctoPrint (0.3.8 and after, can't remember if 0.3.7 also affected) seems broke the iOS App (Ver 3.12 build 4).
Previous version could send notifications like "print progress 25%" or "print complete" normally. After updated OctoPod on OctoPrint, all the notifications are now just "Unknown code".

gdombiak, 2021-08-31:
Hi @paul-tian,

Thanks for reporting this. Could you share your octoprint.log file? It should contain an error for the missing code and the missing language when this error happens. With that info we can troubleshoot and fix this.

BTW, update to version 0.3.9 that has some nice improvements. However, I assume you will see this same error.

Thanks,
Gaston

paul-tian, 2021-08-31:
[octoprint.log](https://github.com/gdombiak/OctoPrint-OctoPod/files/7080660/octoprint.log)
Hello Gaston,

Here is my octoprint.log file. Sure there is something like 
"<timestamp> - octoprint.plugins.octopod - ERROR - Missing translation for code Print progress in language zh".
I assume that the missing language translation is the reason for the "Unknown code".

BTW I could contribute to the translation of "ZH-CN" if needed.

gdombiak, 2021-08-31:
oh, that explains why it's not working. plugin was expecting to see ZH-CN and not zh. That is why it's not finding the correct translation. Is there a way to configure your phone to use zh-cn? or is it always zh?

I might add some lookup logic that if someone looks for zh then it can use zh-cn.

谢谢
Gaston

paul-tian, 2021-08-31:
I've tried to set my phone's region to China Mainland and Language to Simplified Chinese but the log still shows "missing zh". I'm not sure if my phone's model (not a China Mainland Version) is affecting. 

I'll try to test with a China Mainland Version later.

gdombiak, 2021-08-31:
no worries. I'm pushing a new version right now with the proper fix.

gdombiak, 2021-08-31:
Done. Could you update to 0.3.10 release and let me know how things work for you? To force an update go here
[image]

谢谢
Gaston

closed 2021-08-31 by gdombiak

paul-tian, 2021-08-31:
Hello Gaston, 

The 0.3.10 Version of OctoPod for OctoPrint seems good. Test Notification works as expected.

Cheers,
Paul

gdombiak, 2021-08-31:
Excellent!!! Thanks for testing it!

commit 8516bff2 (2021-08-31): Use proper lookup language code for Simplified Chinese and Lithuanian

Closes #122