gdombiak/OctoPrint-OctoPod #139: Plot additional M105 temps in graph - Enhancement

jalanjarosz, 2022-01-30:
Klipper FW allows for additional temperature sensors that can be announced though the standard M105 G-Code.

Could OctoPod temp graph be enhanced to plot additional temperatures reported through M105?

Here is a sample M105 response from my printer:
```
end: M105
Recv: ok B:18.4 /0.0 M4:23.7 /0.0 T0:19.1 /0.0 sbc:47.0 /0.0
Send: M105
Recv: ok B:18.4 /0.0 M4:23.5 /0.0 T0:19.1 /0.0 sbc:46.0 /0.0
```

'M4' and 'sbc' are custom sensors and they could be named anything.

Thanks for this consideration.

gdombiak, 2022-01-30:
This is a great idea. Scheduling for next release. I do not use Klipper so will let you know when I have a version for you to test and confirm that works fine. Ok?

Gaston

jalanjarosz, 2022-01-30:
Awesome.   Yes, I'd be happy to test it out.
Thank you.

James

gdombiak, 2022-01-30:
Hi @jalanjarosz,

Couple of questions:
1. Does Klipper support [M155](https://marlinfw.org/docs/gcode/M155.html) for auto-reporting? 
2. If M155 is supported, does Klipper include these extra sensors in the report?

If M155 is supported and extra sensors are included then I will see how to capture that info and make OctoPod display them in the temp graph.

Thanks,
Gaston

jalanjarosz, 2022-01-30:
It does not, out of the box.  

I see the input is M155 S# where # is seconds. 

What does the Marlin output look like?

jalanjarosz, 2022-02-01:
I spent a bit of time and checked the M155 command. I can easily write a Klipper macro to support this.

jalanjarosz, 2022-02-03:
@gdombiak adding direct support to Klipper might be a better idea?  That would mean your project isn't tied just to Octoprint.

Aren't you accessing the Octoprint API?  Klipper has Moonraner -

https://github.com/Arksine/moonraker

I haven't installed it yet, but I'm going to soon.  Seems the Klipper developers are moving away from Octoprint as Mainsail and Fluidd web UI's are getting more features. 

What's your thoughts?  I'm willing to assist in the testing portion.