gdombiak/OctoPrint-OctoPod #144: False Positives For Possible Thermal Runaway

johnblackwidowjohn, 2022-03-21:
I have my Max Temperature difference set to 12 and I am getting alerts when things are 9 degrees off. Does the log below confirm that it is trigging off of max temperature and not a time setting?

2022-03-21 17:22:07,749 - octoprint.plugins.octopod - WARNING - Possible thermal runaway detected for tool0. Actual 211.77 and Target 220.0

johnblackwidowjohn, 2022-03-21:
After poking around in the code it looks like there is a second setting called thermal_below_target_threshold with a default of 5. The thermal_protection_notifications file uses this setting to determine if there when the temp is below target not thermal_runway_threshold which is what I set to 12. I do not see an option to set thermal_below_target_threshold so maybe this should be an enhancement.

kylane, 2022-04-24:
Im getting the same....

[image]

[image]

trainman419, 2023-01-02:
I just set up OctoPrint and OctoPod on a brand-new Prusa i3, and I'm getting these notifications too.

I'm doing a pretty fast print in PETG, and I see some significant changes in nozzle temperature, with a target of 240C and peaks around 252C.

I'm fairly new to 3d printing and OctoPrint, but I have a decent software background. Should I be concerned about these temperature swings, or is there a setting I can adjust to make the thermal runaway detection in OctoPod less sensitive?