gdombiak/OctoPrint-OctoPod #148: Print complete notification delay cannot be negative

zobi, 2022-05-15:
Hello, my cam is fixed on z axis. When job is not complete i see correctly the printing, but when the job is complete printer go to home position and after the notification « print complete » is send so i can’t see my printing on the final notification.
My delay notification for print complete is set to 0s and i cannot set a negative value…
Is there any solution to send with negative delay ?

[image]

gdombiak, 2022-05-15:
Hi @zobi,

Negative values won't do the effect you are looking for. I could add an improvement so that you can specify a gcode being sent to the printer to get a notification if that would help. A gcode that you add before moving the bed to the front could be used to get a notification.

Would this help?
Gaston

zobi, 2022-05-15:
Oh my god, off course it can solve my issue ! 
Thanks a lot for your help and sorry for my english, i’m french

CookieMuncherUK, 2022-08-12:
If you can add this option it would be great. I also have my cam on the z-axis