gdombiak/OctoPrint-OctoPod #150: Improved alert notifications with details

arekm, 2022-07-07:
Current notifications are quite vague without any details for example:

"Printer bed below specified temperature threshold'

This is feature request about making these detailed. When there is mention of temperature - say what it is.

"Printer bed (20 °C) below specified temperature threshold (XYZ)'

especially that these values are known and logged:

 self._logger.debug(
                    "Print done and bed temp is now below threshold {0}. Actual {1}.".format(threshold_low,
                                                                                             temps[k]['actual']))

This also applies to:
               "bed-cooled": 'Printer bed below specified temperature threshold',
                "bed-warmed": 'Printer bed warmed to specified temperature and duration',
                "tool0-cooled": 'Extruder below specified temperature threshold',
                "tool0-warmed": 'Extruder warmed to specified temperature',
                "thermal-runaway": 'DANGER: Possible thermal runaway detected!'

Not sure if "mmu-event": 'MMU Requires User Assistance', can have more info like what kind of event is that.

gdombiak, 2022-07-07:
This is great feedback @arekm . Thanks for sharing them. Scheduled for next release.

Gaston

arekm, 2022-07-10:
If print time related data is available at "Print complete" event then print time (vs maybe initially estimated time (in time units or just %)) would be nice there, too.

closed 2022-08-07 by gdombiak