gdombiak/OctoPrint-OctoPod #158: Layer Notification Issue IOS

CookieMuncherUK, 2022-08-12:
Hi, I'm having issues with the layer notifications. It all started today I was getting some issues with the App every time it would try to connect or refresh I would see an issue on my OctoDash screen it would glitch out a bit.

So, I decided to uninstall the app on my iPhone and reinstall it. now I get double or triple notifications. I get double on layer 1 and if I set it on the app, for example, layer 5 I would get triple notifications now. 

I get normal notifications on 50% and on completion but I don't know why I'm getting so many on the first layer.

I have tried deleting my printer from the app and removing it. tried restarting my phone.
also tried uninstalling the plugin on octoprint.

i have done the send test notification and that works fine i also removed registered devices from the plugin.

im on the latest update.

Please help me restore this back to a single notification.
[image]
[image]
[image]

CookieMuncherUK, 2022-08-18:
Fixed it. Had to do a complete clean install of octoprint.

closed 2022-08-18 by CookieMuncherUK

moviebrain, 2025-01-26:
I really wish I knew the root cause of this, even though it’s closed. I’d rather not completely build up my octoprint instance again. The problem followed me through a backup and restore (of course it did) but I’d really prefer knowing what I could reach inside of and clean for this one double/triple notification. It’s only the custom layer notifs too, not the percent progress ones like you’ve said.

gdombiak, 2025-01-26:
Hi @moviebrain ,

Could you go to OctoPrint settings -> OctoPod Notifications -> Devices? Make sure that there is only 1 entry for your phone.

moviebrain, 2025-01-26:
> Hi [@moviebrain](https://github.com/moviebrain) ,
> 
> Could you go to OctoPrint settings -> OctoPod Notifications -> Devices? Make sure that there is only 1 entry for your phone.

[image]

Confirmed, only 1 entry for Devices

moviebrain, 2025-01-26:
I also disabled the custom layer notifications (in my case, for Layer 2) and I got zero notifications, so it isn't coming in from a different setting. Upon re-adding a single Layer 2 notification I again got two back to back for Layer 2.

gdombiak, 2025-01-26:
@moviebrain, that's interesting. Could you run 2 experiments and 1 ask? 

Experiment 1: Try printing something [like this](https://www.printables.com/model/4100-extrusion-and-temp-calibration-by-nozzle-size-with). Curious to see if this happens with any simple print.

Experiment 2: If you have other plugins installed, could you disabled them and see if you still get 2 notifications? I wonder if some other plugin is messing up with OctoPrint events.

Ask 1: If previous experiments did not help explain the issue then enable debug logging under OctoPrint settings -> OctoPod Notifications. It's at the bottom of that page. Run another print that sends 2 notifications and then attach here the octoprint.log file. 

Thanks,
Gaston

moviebrain, 2025-01-26:
Absolutely, please allow me a day or two since my work week is starting up again.

moviebrain, 2025-01-27:
OK, disabled all 3rd party plugins except the required DisplayLayerProgress and the OctoPod plugin.

[image]

I restarted Octoprint, reloaded the UI, then started the small print linked above.

The layer notifications in OctoPod plugin are for 50% and Layer 2.

[image]

During the print I did not get any duplicate notifications.

I re-enabled about half of my installed plug-ins, again did not get any duplicate notifications.

I re-enabled the remaining three 3rd party plugins installed...

[image]

... and no more duplicate notifications.

Thank you very much for your time, and i wish I knew the answer to this puzzle, but at least I have a solution. Disable all 3rd party plugins, restart/reload, start a small print, then re-enable all plugins. Unsure if starting a new print is needed but it's what I've got. 

Thanks again.

moviebrain, 2025-02-01:
I'm sorry, apparently my fix was temporary.

I'm attaching the debug log you requested in a previous comment, and I see both layer notifications in there back to back:

2025-01-31 10:20:41,941 - octoprint.plugins.octopod - DEBUG - Snapshot URL: http://localhost:8080/?action=snapshot 
2025-01-31 10:20:42,426 - octoprint.plugins.octopod - DEBUG - Sending notification for event 'layer_changed' (OctoPrint instance on octopi4)
2025-01-31 10:20:43,079 - octoprint.plugins.octopod - DEBUG - Notification Response code: 204
2025-01-31 10:20:43,099 - octoprint.plugins.octopod - DEBUG - Snapshot URL: http://localhost:8080/?action=snapshot 
2025-01-31 10:20:43,539 - octoprint.plugins.octopod - DEBUG - Sending notification for event 'layer_changed' (OctoPrint instance on octopi4)
2025-01-31 10:20:44,089 - octoprint.plugins.octopod - DEBUG - Notification Response code: 204

[octoprint(2).log](https://github.com/user-attachments/files/18626770/octoprint.2.log)

moviebrain, 2025-03-09:
Sorry to poke this again. The layer 2 issue is ongoing for me. Is there any other debug or information I can provide?