gdombiak/OctoPrint-OctoPod #172: Notifications don't seem to be working

gary-wlg, 2023-10-16:
It appears that notifications are not working, I receive a send failed when trying to test the notifications.

gdombiak, 2023-10-16:
Indeed it's not working again. Today I completely rebuilt the service and was working well ...... for a few hours it seems. :( Looking to see what's going on.

Thanks,
Gaston

gary-wlg, 2023-10-16:
Thanks Gaston. I did notice that it didn't hang OctoPrint anymore, it just gave the error quickly and continued on - which is a definite improvement :)

gdombiak, 2023-10-16:
Honestly, we got lucky! This is another kind of error that luckily does not block OctoPrint.

gary-wlg, 2023-10-16:
I'll take luck any day!  I have 30 printers running this, I don't think I could live without these notifications :)

gdombiak, 2023-10-16:
And we are back in business!!! One of the services was not correctly restarting when AWS instance was rebooted. All good now.

Thanks again for letting me know about this.
Gaston

closed 2023-10-16 by gdombiak