gdombiak/OctoPrint-OctoPod #173: Can't install on Pi3B

MarioBerger, 2023-10-22:
Hello,

ich try to install the Plugin.
Over the PlugInManager I have following feedback
```
Installiere Plugin "OctoPrint-OctoPod" von https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip...
Fehler!
Unexpected error while trying to install plugin from None
```

And over commandline this:

```
mario@octopi:~ $ pip install "https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip
  Downloading https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip
     / 55 kB 1.4 MB/s
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-_n6g3jjt/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-_n6g3jjt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-bvjaixbp
         cwd: /tmp/pip-req-build-_n6g3jjt/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
WARNING: Discarding https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip. Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.

```

Octoprint is a new installation from yesteday.

gdombiak, 2023-10-22:
Hi Mario,

I think that you are facing 2 different issues in your post. 1. The one from OctoPrint is the real one and will get to it in a bit and 2. the one from the command line.

To run pip from the command line you first need to make sure you are using the correct python virtual environment. Once you are into the proper python environment, you can then run pip and you should hit the same error you see from OctoPrint but probably with more logging information which will help us troubleshoot the problem.

Regarding the issue from OctoPrint, the logging information is minimal and checking OctoPrint logs may throw some more light into this. With that limited information I found these links that point to a network connectivity issue:

Link 1: https://www.reddit.com/r/octoprint/comments/13ujkjk/needing_help_installing_plugins/
Link 2: https://community.octoprint.org/t/octopi-cant-connect-to-internet-with-wireless-bridge/52529

In summary, try running pip from the proper environment to gather more logging information or check OctoPrint's logs. Maybe you do have a connectivity issue but hard to tell.

Regards,
Gaston

MarioBerger, 2023-10-23:
Hello and thank you for the fast answere.

My Octopi has no problems with internet connections.
```
mario@octopi:~ $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=119 time=17.4 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=119 time=19.1 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=119 time=18.2 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=119 time=17.7 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=119 time=15.2 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=119 time=14.9 ms
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5008ms
rtt min/avg/max/mdev = 14.887/17.080/19.139/1.540 ms
```
And I have installed a Desktop Plugin normaly over the Plugin Manager.
pip is installed 
But I run currently a apt upgrade and he has a lot to do.
I will try it again after, but now my work calls.

MarioBerger, 2023-10-23:
Ok, after a apt update and upgrade I can install it with the PlugIn Manager.
I'm sorry for this dump failure from me.

closed 2023-10-23 by MarioBerger