gdombiak/OctoPrint-OctoPod #178: Running notification test causes a crash

SeriousBug, 2023-12-19:
Hey! Using the OctoPi distribution, I'm experiencing a crash whenever I press the "Send test notification" button. I haven't tested if it happens with an actual notification as well. It doesn't seem to crash OctoPrint itself, but I had to SSH in and restart the haproxy service. It happens consistently whenever I press the button. Here are a few relevant snippet from OctoPrint logs:

```
2023-12-19 15:30:49,843 - octoprint.plugins.octopod - WARNING - Live Activity Notification Response: b'<!doctype html>\n<html lang=en>\n<title>404 Not Found</title>\n<h1>Not Found</h1>\n<p>The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.</p>\n'
```

```
2023-12-19 15:30:49,843 - octoprint.plugins.octopod - WARNING - Live Activity Notification Response: b'<!doctype html>\n<html lang=en>\n<title>404 Not Found</title>\n<h1>Not Found</h1>\n<p>The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.</p>\n'
```

```
2023-12-19 16:02:08,292 - octoprint.plugins.octopod - INFO - Could not send Live Activity Notification: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
2023-12-19 16:02:08,293 - octoprint.plugins.octopod - DEBUG - Live activity - Activities: 1, Priority: 5, Progress: 6, State: Printing and Time Left: 10726
2023-12-19 16:02:08,302 - octoprint.plugins.octopod - WARNING - Could not send message: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
2023-12-19 16:02:08,309 - octoprint.plugins.octopod - INFO - Could not send Live Activity Notification: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
2023-12-19 16:02:08,312 - octoprint.plugins.octopod - INFO - Could not send Live Activity Notification: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
2023-12-19 16:02:08,313 - octoprint.plugins.octopod - INFO - Could not send Live Activity Notification: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
2023-12-19 16:02:08,316 - octoprint.plugins.octopod - DEBUG - Live activity - Activities: 1, Priority: 5, Progress: 3, State: Printing and Time Left: None
2023-12-19 16:02:08,312 - octoprint.plugins.octopod - INFO - Could not send Live Activity Notification: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
```
Full logs: [octoprint.log](https://github.com/gdombiak/OctoPrint-OctoPod/files/13720854/octoprint.log)

SeriousBug, 2023-12-19:
Also, oh wow OctoPod prints out so many debug logs. 23224 of 25732 lines in the logs is just OctoPod.

gdombiak, 2023-12-20:
Hi @SeriousBug ,

Could you paste here the exact "Notification Server URL" you have in the OctoPod Plugin setting page?

Thanks,
Gaston

SeriousBug, 2023-12-21:
@gdombiak It's `http://100.87.49.23`.

gdombiak, 2023-12-21:
Hi @SeriousBug ,

Thanks for sharing that information. The "Notification Server URL" should point to http://octopodprint.com/ which is the public server for sending iOS notifications. Update that field and let me know how it goes.

Thanks,
Gaston

SeriousBug, 2023-12-21:
Thank you! Fixing the Notification Server URL solved the issue.

closed 2023-12-21 by SeriousBug