gdombiak/OctoPrint-OctoPod #181: Running notification test fails

lemming622, 2024-04-03:
While this sounds like a duplicate of #178, I have confirmed that the *Notification Server URL* is in fact ```http://octopodprint.com/```.  

When I *Send test Notification* I get prompted with ```Sending failed: There was an error sending message. Check logs```

Checking the logs gives me ```2024-04-02 18:40:18,396 - octoprint.plugins.octopod - WARNING - Could not send message: HTTPConnectionPool(host='octopodprint.com', port=80): Max retries exceeded with url: //v1/push_printer (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71b470d0>: Failed to establish a new connection: [Errno 101] Network is unreachable'))```

Doing a site site uptime looks like the site is down.  (https://www.isitdownrightnow.com/octopodprint.com.html)

Should there be another address to use or am I doing something wrong?

gdombiak, 2024-04-03:
Hi @lemming622,

Amazon will start charging more for using IPv4 so today I changed the push notification service to only run on IPv6. I just reverted the change so both IPv4 and IPv6 are available. Give it a few hours and try again. Let me know if things work again. 

Thanks,
Gaston

lemming622, 2024-04-03:
works now.  Thank you

closed 2024-04-03 by lemming622

gdombiak, 2024-04-03:
Excellent. Thanks for reporting back. Will keep ipv4 enabled then since ipv6 is not fully supported out in the wild. Bummer Amazon cost will go up. Thanks and sorry for this