gdombiak/OctoPrint-OctoPod #194: OctoPod plugin blocks printing when service is down

mkevenaar, 2025-03-16:
```
2025-03-16 20:49:58,180 - octoprint.plugins.octopod - WARNING - Could not send message: HTTPConnectionPool(host='octopodprint.com', port=80): Max retries exceeded with url: /v1/push_printer (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x9b688898>: Failed to establish a new connection: [Errno 101] Network is unreachable'))
```

With the plugin enabled, I currently cannot print

macochran, 2025-03-17:
Is the service down for you too?  I stopped getting notifications yesterday for some reason, and I've tweaked it for hours and can't get it to send me a notification.

mkevenaar, 2025-03-17:
> Is the service down for you too? I stopped getting notifications yesterday for some reason, and I've tweaked it for hours and can't get it to send me a notification.

Yes, I am unable to get notifications. But when I try to start a print, it just sits and waits until it can send that notification. I have disabled the plugin.

macochran, 2025-03-17:
Interesting.  Mine still prints just fine, but I can't get it to send any notifications at all.

MateMago, 2025-03-18:
Mine prints fine, but I have to wait 2 minutes before Prusa MMU Plugin prompts me to select filament in single mode. I disabled OctoPod and now it prompts immediately. No notifications from Octopod though.
I believe if you just wait 2-3 mins the print will eventually start, but that's annoying.

gdombiak, 2025-03-18:
Thanks guys for letting me know the server went down. I just rebooted it and it's working again. Since running this server costs money, I'm using the smallest possible machine and I think we ran out of memory. If this happens again then a bigger machine will be needed which implies more $$$. AWS does not have free-tiers for open source projects. :(

macochran, 2025-03-18:
Thank you so much for the update, and for continuing this service. I absolutely appreciate it. 

Is there a way for us to check the status of the server in the future while we are troubleshooting ?  I thought I screwed something up on my end. 😃

gdombiak, 2025-03-19:
good idea @macochran . I will see of adding a status to the octopod settings page to make it easier to see what is going on. Will keep this ticket opened and use it for this improvement.

macochran, 2025-03-19:
Thanks again, and thanks for a great plugin!